OctoPrint/OctoPrint #2729: [Request] Allow additional video formats to appear in the timelapse tab

FormerLurker, 2018-07-12:
Is it possible to add support for the following formats within the timelapse file browser:
GIF
AVI
FLV
VOB
MPEG (might already work actually, need to test)

As you may have guessed, this is for Octolapse so that I can synchronize additional formats that are supported by the plugin.  Currently I can copy the files into the timelapse folder, but they will not display and I cannot download them with a direct link.

Thanks!

#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

koenkooi, 2018-07-13:
I added some formats a while ago, have a look at https://github.com/foosel/OctoPrint/pull/2409 to see the work involved.

FWIW, 'vob' is currently the default format, but using the 'mpg' suffix.

foosel, 2018-07-13:
I just added a new hook [`octoprint.timelapse.extensions`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-timelapse-extensions) that can be used to add additional supported extensions.

Available on `maintenance`, soon `devel`, to be released with 1.3.10.

FormerLurker, 2018-07-13:
@foosel,  the hook worked as expected, excellent!  👍  This is a pretty big upgrade for Octolapse since the other video formats were too much of a bother to use before.

I did have to remove the _sendingLock from the _handle_ok method to test it out, but I was expecting that.  Now to remove all of the warnings about which formats will sync from my settings and help files.

foosel, 2018-12-10:
1.3.10 has been released.

closed 2018-12-10 by foosel