OctoPrint/OctoPrint #2735: Plugins wont install

ShishaTobi, 2018-07-15:
I try to install the Telegram Plugin but it just fails.
I dont know why...

here the log:
2018-07-15 17:02:36,705 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/master.zip
2018-07-15 17:02:36,783 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-07-15 17:02:37,694 - octoprint.plugins.pluginmanager - ERROR - Could not install plugin from https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/master.zip
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 348, in command_install
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 652, in _call_pip
    return self._pip_caller.execute(*args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 172, in execute
    return self.call(command, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/commandline.py", line 83, in call
    p = sarge.run(command, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
TypeError: __init__() got an unexpected keyword argument 'async'
2018-07-15 17:02:37,700 - tornado.access - ERROR - 500 POST /api/plugin/pluginmanager (192.168.178.46) 1014.40ms
2018-07-15 17:02:43,652 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.178.46

Can someone please help me?
Thank you.

ntoff, 2018-07-16:
Looks like the same issue as https://github.com/foosel/OctoPrint/issues/2687

foosel, 2018-07-16:
Yep, duplicate. Added [an FAQ entry for this](https://discourse.octoprint.org/t/i-cant-install-uninstall-a-plugin-i-get-something-about-an-unexpected-keyword-argument-async/2801). Closing.

closed 2018-07-16 by foosel