OctoPrint/OctoPrint #2737: Print button prematurely enabled

OutsourcedGuru, 2018-07-17:
#### What were you doing?
Opened up OctoPrint, (auto-connected to printer) and noticed that the Print button is enabled. I have not selected a file yet to print.

#### What did you expect to happen?
I seem to recall that in earlier versions, the Print button starts off in a disabled state.

#### What happened instead?
The button seems to be available prematurely.

#### Version of OctoPrint
1.3.9rc2

#### Operating System running OctoPrint
OctoPi (Raspian Stretch) 0.15.0

#### Printer model & used firmware incl. version
Robo 3D - Robo C2

```
FIRMWARE_NAME:Marlin 1.1.7-C2 (Github) SOURCE_CODE_URL:https://github.com/Robo3D/Marlin-C2 PROTOCOL_VERSION:C2 MACHINE_TYPE:RoboC2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff EMERGENCY_CODES:M108,M112,M410
```

#### Browser and version of browser, operating system running browser
Safari 11.1

#### Link to octoprint.log
[octoprint-2.log](https://github.com/foosel/OctoPrint/files/2202767/octoprint-2.log)

#### Screenshot(s)/video(s) showing the problem:
[image]

foosel, 2018-07-18:
Thanks for spotting this regression, will be fixed in 1.3.9rc4

OutsourcedGuru, 2018-07-18:
I'm on 1.3.9rc3 now, for what it's worth (same behavior).  Thanks.

foosel, 2018-07-20:
1.3.9rc4 is out now.

closed 2018-07-20 by foosel

OutsourcedGuru, 2018-07-23:
I can confirm that it works in 1.3.9rc4. Thanks.

OutsourcedGuru, 2018-07-25:
@foosel This didn't make it back into the full release of 1.3.9, did it?

[image]

foosel, 2018-07-26:
That is a disabled print button. An enabled one isn't flat:

[image]

OutsourcedGuru, 2018-07-26:
Ah, subtle it is.  Thanks.

commit 2062acd3 (2018-07-18): Fix print button being enabled w/o selected file

Closes #2737