OctoPrint/OctoPrint #2743: Triggering rpicam by a gcode

rafaa1988, 2018-07-19:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-07-19:
?

closed 2018-07-19 by foosel

rafaa1988, 2018-07-19:
Sorry, I'm newbie. I simply want to let the picam capture image on z change and or on gcode in octoprint. I found in Octoprint timelapse recording or image capturing but this is not what i need. As an example the m300 works for sound. Is there a specific gcode for triggering the camera? and is z change implemented in octoprint. Thanks.