OctoPrint/OctoPrint #2746: Malicious link inside Cura installation tutorial

eyasukochi, 2018-07-21:
#### What were you doing?

1. I was exploring the tutorial docs on how to install Cura
2. I wanted to see more about a particular pre-built Raspbian image
3. I clicked a link going to http://bit.ly/octopi_cura_engine_150406 which is inside of the "Compiling for Raspbian" note section

#### What did you expect to happen?

I expected to get an image download, or to be redirected to a file directory

#### What happened instead?

I went through what looked like way too many redirects, and then was shown a dialog insisting I install flash, which wouldn't allow me to exit out of and when I clicked "later" it began a download of a dmg (I'm on OSX) file anyway. This seems unlikely to be the intended result you folks had. I think your hosting site has been compromised.

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPrint

N/A

#### Operating System running OctoPrint

N/A

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ. YES

ntoff, 2018-07-21:
Which  specific tutorial doc were you following?

eyasukochi, 2018-07-21:
yes that's the section. There's a bit.ly link in there that redirects somewhere suspicious.

hashashin, 2018-07-22:
Most likely someone "clever" making profit of the customization link thing in bit.ly. I guest the actual one expired or something.

hashashin, 2018-07-22:
I just discovered that if you add "+" at the end of the link bit.ly will show some info instead of redirect. 
https://bitly.com/octopi_cura_engine_150406+
Nothing of interest however.

foosel, 2018-07-23:
Huh... Looks like there was a typo. It was supposed to be `http://bit.ly/octopi_cura_engine_150406`, so **0406** instead of **0604**. Sorry for that, thanks for the heads-up and the PR (which I'll merge to `master` actually to make sure the docs get updated immediately).

foosel, 2018-07-23:
Patch is live at http://docs.octoprint.org/en/master/bundledplugins/cura.html, closing this. Thanks again!

closed 2018-07-23 by foosel