OctoPrint/OctoPrint #2752: G33 P10 broken with latest RC

jasonkfirth, 2018-07-23:
I updated to the latest Octoprint RC and immediately ran a G33 P10 to auto calibrate.

It seems like octoprint keeps trying to get temperatures during the calibration, throwing a bunch of timeout errors in the terminal. After the calibration is complete, the printer thinks it has paused a print.

Using a FLSun Delta Kossel running the latest Marlin

foosel, 2018-07-24:
You sadly didn't provide any of the logs I need for further analysis. However, based on this:

> It seems like octoprint keeps trying to get temperatures during the calibration, throwing a bunch of timeout errors in the terminal.

it sounds like the firmware isn't adhering to the busy protocol (= not sending `busy: processing` while it can't receive new commands) and you also haven't added `G33` to the list of long running commands. So there's nothing telling OctoPrint that the printer is in fact currently busy and from OctoPrint's perspective it has simply gone AWOL and refuses to talk to it now, which usually is not caused by commands like this but rather serious communication issues which OctoPrint tries to work around by attempting to trigger the printer into talking to it again (the temperature queries you are seeing).

In short: this has to be fixed either in the firmware (that would be the clean solution) or in OctoPrint's serial settings (to work around this problem). It can't be solved in a change of OctoPrint's general behaviour here, since that would cause *severe* issues in case of actual communication loss.

foosel, 2018-07-25:
It may be a change to the worse, but *as far as I can tell right now* it is not caused by something that OctoPrint does wrong but rather a firmware misbehaviour or misconfiguration that so far went undetected.

I have to make *some* assumptions regarding talking to the thousands of firmware variants out there, and I can't work around every quirk of every single one. If I tried, I could just stop working on this now because I wouldn't be able to do anything anymore - a workaround for one variant could break another.

When will you get back to your printer to complete the ticket with logs and such? This RC has now been in RC mode for well over a month with so far only positive feedback apart from this issue and I actually wanted to release today. I'd hate to have to shift this release out until next week (as a rule I no longer release later than Wednesdays since *IF* something goes wrong fixing that will then run into the weekend).

jasonkfirth, 2018-07-25:
If nobody else is having the same problem and nobody else can reproduce (ie. the latest rc doesn't break g33 for anyone else), then unfortunately there's nothing to be done, I suppose.

I'm far away from my printer until Friday, no sense holding the whole world up as long as it's not a problem that affects everyone.

foosel, 2018-07-25:
Ok... Since I haven't heard from anyone else about this, let's tackle this Friday/next week then and if push comes to shove I'll have to push a hotfix release if it really ends up to be an issue with OctoPrint that can only be worked around with an actual code change.

foosel, 2018-09-21:
Apologies. I had a situation in my private life that needed handling and am still working through the backlog from what came up during that...

So, it looks like the issue here is that the firmware sends a `busy` message, OctoPrint detects that, tries to set the new busy timeout, but can't because it can't get the command through to the printer yet (of course not, it's busy), and that leads to OctoPrint thinking the timeout should be a different (smaller) than it is. I have an idea on how to fix this, however it should be possible to work around the issue by simply disabling explicit busy protocol support by unchecking Settings > Serial > Firmware & protocol > Firmware specific settings > Advanced options > Support `busy` protocol, if detected.

[image]

foosel, 2018-09-21:
Should be fixed in 1.3.10 - at least right now I can't reproduce the behaviour against a correspondingly prepped virtual printer.

closed 2018-12-10 by foosel

jasonkfirth, 2018-12-28:
Confirmed that the latest update solves the problem. Thanks!