OctoPrint/OctoPrint #2753: Settings -> Close button inoperative

OutsourcedGuru, 2018-07-23:
#### What were you doing?
I was using the `Settings` dialog to adjust something. When I'd finished SAVE'ing, I then pressed the `Close` button which didn't work.

#### What did you expect to happen?
I expected the `Close` button to dismiss the `Settings` dialog in a similar way that the x in the upper/right corner does.

#### What happened instead?
Nothing appears to happen.

#### Did the same happen when running OctoPrint in safe mode?
No. The `Close` button behaves as expected, dismissing the dialog.

#### Version of OctoPrint
1.3.9rc4

#### Operating System running OctoPrint
OctoPi 0.15.0

#### Printer model & used firmware incl. version
Robo 3D -> Robo C2

```
FIRMWARE_NAME:Marlin 1.1.7-C2 (Github) SOURCE_CODE_URL:https://github.com/Robo3D/Marlin-C2 PROTOCOL_VERSION:C2 MACHINE_TYPE:RoboC2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff EMERGENCY_CODES:M108,M112,M410
```

#### Browser and version of browser, operating system running browser
Safari 11.1, OSX 10.13.4

#### Link to octoprint.log

```
2018-07-23 09:12:28,271 - octoprint.startup - INFO - ******************************************************************************
2018-07-23 09:12:28,276 - octoprint.startup - INFO - Starting OctoPrint 1.3.9rc4
2018-07-23 09:12:28,277 - octoprint.startup - INFO - ******************************************************************************
2018-07-23 09:12:29,518 - octoprint.startup - INFO - Blacklist processing done
2018-07-23 09:12:29,606 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-23 09:12:34,056 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 22 hook handlers
2018-07-23 09:12:34,355 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-07-23 09:12:34,423 - octoprint.server - INFO - Intermediary server started
2018-07-23 09:12:34,424 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-23 09:12:35,881 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 22 hook handlers
2018-07-23 09:12:35,967 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-07-23 09:12:37,717 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-07-23 09:12:37,728 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2018-07-23 09:12:38,097 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-07-23 09:12:42,453 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-07-23 09:12:46,474 - octoprint.util.pip - INFO - Version of pip is 10.0.0
2018-07-23 09:12:46,477 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-07-23 09:12:46,478 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-07-23 09:12:46,499 - octoprint.plugin.core - INFO - Initialized 18 plugin implementation(s)
2018-07-23 09:12:46,539 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO801 = /usr/bin/aplay /home/pi/sounds/r2d2/1.wav &
2018-07-23 09:12:46,541 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO802 = /usr/bin/aplay /home/pi/sounds/r2d2/2.wav &
2018-07-23 09:12:46,543 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO803 = /usr/bin/aplay /home/pi/sounds/r2d2/3.wav &
2018-07-23 09:12:46,547 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO804 = /usr/bin/aplay /home/pi/sounds/r2d2/4.wav &
2018-07-23 09:12:46,548 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO805 = /usr/bin/aplay /home/pi/sounds/r2d2/5.wav &
2018-07-23 09:12:46,549 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO806 = /usr/bin/aplay /home/pi/sounds/r2d2/6.wav &
2018-07-23 09:12:46,550 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO900 = /usr/bin/ssh -t pi@pisense.local "sudo pkill temp-graph.py && sudo /home/pi/scripts/cls.py && sleep 3 && sudo poweroff"
2018-07-23 09:12:46,556 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO902 = /usr/bin/ssh -t pi@picam1.local "sudo pkill mjpg_streamer && sleep 3 && sudo poweroff"
2018-07-23 09:12:46,585 - octoprint.plugin.core - INFO - 18 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Bed Leveling Wizard (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelingwizard
|  Bed Visualizer (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  EEPROM Marlin Editor Plugin (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  GCODE System Commands (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_gcodesystemcommands
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Navbar Temperature Plugin (0.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
[... 159 lines cut ...]

#### Link to contents of Javascript console in the browser

Home page before bringing up the `Settings` dialog:
```
Could not bind view model – "PrintTimeGeniusViewModel" – "to target" – "#settings_plugin_PrintTimeGenius" – ":" – "text↵update@http://charming-pascal.local/static/webassets/packed_libs.js?331a1bb5:11404:23…"
error
  (anonymous function) — packed_core.js:15072
  Pn — packed_libs.js:11208:532
  (anonymous function) — packed_libs.js:11229
  (anonymous function) — packed_core.js:15035
  Pn — packed_libs.js:11208:532
  (anonymous function) — packed_libs.js:11229
  bindViewModels — packed_core.js:15002
```

After bringing up the `Settings` dialog (at the default `Serial Connection` tab) and pressing `Close`:
```
ReferenceError: Unable to process binding "text: function (){return firstFilamentPrintTime().toFixed(1) }"
Message: Can't find variable: firstFilamentPrintTime
```

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

OutsourcedGuru, 2018-07-23:
I used a binary search to find the offending plugin:  PrintTimeGenius. Disabling the plugin makes the issue go away.

I'll recreate the issue over there on his repository. Feel free to close this, having noted the behavior.

foosel, 2018-07-24:
That's why I always ask people to test in safe mode - if an issue goes away in safe mode, a third party plugin is at fault :)

Closing this since you already identified the culprit 🎉

closed 2018-07-24 by foosel

OutsourcedGuru, 2018-07-24:
I mostly just document here (as well as there) so that when people search _here_ then they have somewhere to land.