OctoPrint/OctoPrint #2757: Open Discussion: OctoPrint's 3rd Party Plugin's Privacy Statement.

Thisismydigitalself, 2018-07-25:
Hi All. 
I don't know where else I should post this and who manages OP's 3rd party plugin repository. if you think this belongs elsewhere - let me know. 

Back to Topic: It seems that unlike many others, I have privacy concerns and that's due to this simple fact that this is the 21st century. I will start by saying and I believe I speak on behalf of many if not all others that we are thankful for all of you guys great open source software and plugins so BIG Thank you! goes to all of you. Having said that and looking on how long the list of 3rd party plugins grows each passing day, My wish it to have, much like with Google Play or App Store, a way to know what information OctoPrint's 3rd party plugins collects, if any. I believe we all deserves to know if any of our private data is being collected and shared without our explicit approval. I am not a programmer nor knows how to read and review codes so I would like all those who write plugins for OctoPrint to clearly state what they are collecting, again - if at all - I really don't know. If I missed that and each plug is published with a clear and easy to find privacy statement, please come forward.

I believe my request is basic.
Thank you all.

chatrat12, 2018-07-28:
It's a simply request in theory, I don't think it's so simple in practice. A lot of sandboxing would have to be done and alot of UX design would have to go into such a system. Keeping track of what permissions each plugin has could be a laborious task. And you would have to stop one plugin from spoofing another plugin to gain access permissions that the plugin it is spoofing has already been granted. I personally believe such a system would be no easy feat and al ot about how plugins are packaged/distributed might have to change in order to accommodate the system. It would be great to have, but I wouldn't hold your breath.

I think as a community, we should expect that any plugin author who is collecting our data is transparent about it and makes it known to the end user.

Plugin security is an interesting subject. You are currently at the mercy of the author of the plugin you have installed. There isn't really anything to stop a plugin that improperly collect user data or is just down right malicious. The only remedy is public awareness and its removal from the plugin repo.

Thisismydigitalself, 2018-07-29:
If you ask me, The most difficult hurdle here is to convince others that we do not and will not accept codes that invades our lives. I am not saying they can't chose to go this way but we are entitled to KNOW, to be TOLD, and to say NO-THANK YOU before installing any piece of sw that collects data. My fear is that many do not even care about this privacy issue. 

My wish is for those who are responsible for law and order to demand from Plugin writers to clearly state what they are collecting before they are allowed to publish their codes on OctoPrint's 3rd party plugin Repository. I will continue my struggle until i see serious attempt to put things in order. I believe we all deserves this but it's more than that - it's the law.

https://www.eugdpr.org/

chatrat12, 2018-07-29:
I definitely agree and think we should hold plugin authors to that standard. Plugins don't really have a review process so it's up to the community to keep the honest. Do you know of any plugins that are not disclosing the tracking of information?

jneilliii, 2018-07-29:
I'm not sure if that's true @chatrat12. Every plugin I submit to the repo had to be accepted buy a pull request, so it is being reviewed, just not sure if it is being reviewed for that specific privacy concern.

chatrat12, 2018-07-29:
Yeah, I've done the same but it's not like @foosel can comb through our code to make sure that it's not doing anything improper.

chatrat12, 2018-07-29:
Especially considering that updates to the plugin don't go through any review process. They go straight to the end user.

hashashin, 2018-07-30:
The only plugin that collects data I can see in the repo, it's one not approved yet https://github.com/OctoPrint/plugins.octoprint.org/pull/214 because of that... so @foosel it's concerned about this. 
Perhaps putting a section/warning in the plugins tutorial about making an opt-in thing for plugins that collect data could be enough?

Salud.

Thisismydigitalself, 2018-07-30:
@hashashin, "The only plugin.. ",
Trust me there are more. it's all down to what data would you consider private.

@chatrat12, "I have no doubt the community will address it", 
I am sorry to say that from my experience, most users just install things without any privacy concerns whatsoever - none. these days people are installing spyware in the form of cool App for Android without giving any thoughts.

Scary.

foosel, 2018-08-01:
Sorry, it took me a bit to chime in here. But maybe that's actually a good thing since now you all had a chance to discuss a bit already without me weighing in and potentially influencing things ;)

> My wish is for those who are responsible for law and order to demand from Plugin writers to clearly state what they are collecting before they are allowed to publish their codes on OctoPrint's 3rd party plugin Repository.

I added this to the submission guidelines a while ago:

[image]

With that being said, what was already pointed out is correct - me or @kantlivelong *do* take a brief look at all plugins that are submitted before merging, but we don't have the resources for a full blown review of the code and any updates that might come later.

There is indeed abuse potential here, and that bothers me, but with the available resources it's impossible to do anything else. I have to rely on the community crying foul if some plugin is discovered that is collecting data without saying, so it can be addressed and if the author should proof uncooperative also be removed from the repository for good.

Apart from me there is no one else working on OctoPrint full time (AFAIK), so a review system like seen on the phone stores is simply out of the question. And the nature of the plugin system and the underlying Python runtime also would make actual sandboxing extremely tricky, maybe even impossible, and in any case mean the end of the current and the creation of a new plugin system and possibly underlying full system architecture with it - and correspondingly all existing plugins would become incompatible. In short, it would be a Mons Olympus of tasks that's absolutely possible to do with the way things are.

Thisismydigitalself, 2018-08-05:
Is there a tool/software for users like myself for, as much as possible, automate code review for finding privacy breaches?

chatrat12, 2018-08-05:
Automated code review software exists but it's usually not for security purposes. In a non sandboxed environment, I believe it would be very difficult for any type of software to discriminate code that is transmitting data. 

You could monitor outbound traffic, that would be pretty difficult to analyze and seems like a dead end to me.

For ultimate security, your best solution is to just take your device offline and install plugins via zip files. I think you can still have it on the local network and disable its internet access by giving it a phony DNS although, I have not tried.

foosel, 2018-08-06:
Believe me, if there was a way to automatically perform code reviews/tracking audits, I would use it. 

Isolation on the network might indeed be your best bet here.

Thisismydigitalself, 2018-08-07:
@foosel , Can you please demand from plugin's writers to pop up a window clearly stating what data, if at all, will be collected BEFORE we install any plug from OP 3rd party plugin repository? not vague info - clear and accurate information about what their plug collects. I hate prolonging this thread longer than needed but i truly am committed to see progress and a change for the better when it come to plugins or any piece of software I install. I hate pushing this subject as I am speaking for myself and I do not know how many feels the same like I do and hate not knowing what goes on under the hood. please don't let this subject fade out.

Thank you.

foosel, 2018-08-07:
I am already demanding that they provide this kind of information in the plugin description and do not perform any tracking at all until the user has opted-in. I can't do much but this (and de-listing plugins that are found to not adhere to this requirement, for which I need the community's help though because I can't review each and every plugin on each and every update).

chatrat12, 2018-08-07:
> Can you please demand from plugin's writers to pop up a window clearly stating what data, if at all, will be collected BEFORE we install any plug from OP 3rd party plugin repository?

Plugins get via installed pip, I don't think there is anyway for it get that kind of info before installing the plugin, however, I'm not a python expert so I could be wrong. As far as I understand, this is just how the software distribution ecosystem works in these environments.

> i truly am committed to see progress and a change for the better when it come to plugins or any piece of software I install.

I agree that everyone should value their privacy. As far as progress and a change for the better, have there been plugins out there that have been abusing user data? Even if someone were to scrape data from my Pi, as long as you use a VPN, there isn't a ton of info for them to grab. They can see my printing habits and that is about it. I feel like there are far more pressing devices we need to guard our privacy with. Personal computers and phones are ripe with personal data and that data is being abused all the time. I don't know of any personal data being abused by plugins on OctoPrint. Your original request requires a ton of resources for something that really hasn't been a problem so far.