OctoPrint/OctoPrint #2758: Disconnected from Pursa i3 Steel (MKS Gen 1.4) after extruders heating start

rashn, 2018-07-25:
#### What were you doing?

1. Open STL file
2. Start printing
3. Bed reached 60 C degree. 
4. Start heating extruders (dual extruder MK8)
5. Serial connection lost
6. Printer still heating and stays with correct temperature.

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

In safe mode same trouble.

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

Raspberry Pi 3
Linux octopi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

#### Printer model & used firmware incl. version

Prusa i3 Steel dual extruders MK8 (clone)
Printer firmware (Marlin 1.1.8)

#### Browser and version of browser, operating system running browser

Mac Os High Sierra
Chrome

#### Link to octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/2228840/octoprint.log)

#### Link to contents of terminal tab or serial.log

[serial.log](https://github.com/foosel/OctoPrint/files/2228841/serial.log)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
[image]

[sensor2.gcode.zip](https://github.com/foosel/OctoPrint/files/2228844/sensor2.gcode.zip)

I have read the FAQ.

rashn, 2018-07-25:
Another attempt, screenshot:
[image]

foosel, 2018-07-26:
Based on your `serial.log`, your printer does support the busy protocol in order to tell the host every 2s that it's still alive even if blocked by some command. You can see this in action in the handling of the `M190` command where it works flawlessly - the printer performs a blocking heatup, keeps reporting the temperatures and also sends an `echo:busy: processing` every 2s:

```
2018-07-25 16:50:38,796 - Send: N5 M190 S60*91
2018-07-25 16:50:38,805 - Recv:  T:56.59 /0.00 (873.38) B:29.77 /60.00 (966.50) T0:56.59 /0.00 (873.38) T1:37.08 /0.00 (947.75) @:0 B@:0 @0:0 @1:0 W:?
2018-07-25 16:50:38,915 - Recv:  T:56.58 /0.00 (873.44) B:29.69 /60.00 (966.69) T0:56.58 /0.00 (873.44) T1:37.06 /0.00 (947.81) @:0 B@:127 @0:0 @1:0
2018-07-25 16:50:39,806 - Recv:  T:56.42 /0.00 (874.19) B:29.72 /60.00 (966.63) T0:56.42 /0.00 (874.19) T1:37.15 /0.00 (947.56) @:0 B@:127 @0:0 @1:0 W:?
2018-07-25 16:50:40,805 - Recv:  T:56.28 /0.00 (874.88) B:30.16 /60.00 (965.63) T0:56.28 /0.00 (874.88) T1:37.17 /0.00 (947.50) @:0 B@:127 @0:0 @1:0 W:?
2018-07-25 16:50:40,809 - Recv: echo:busy: processing
2018-07-25 16:50:40,816 - Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
2018-07-25 16:50:41,806 - Recv:  T:56.12 /0.00 (875.63) B:30.36 /60.00 (965.13) T0:56.12 /0.00 (875.63) T1:36.79 /0.00 (948.63) @:0 B@:127 @0:0 @1:0 W:?
[...]
2018-07-25 16:51:54,805 - Recv: echo:busy: processing
2018-07-25 16:51:55,800 - Recv:  T:48.63 /0.00 (908.19) B:60.16 /60.00 (856.13) T0:48.63 /0.00 (908.19) T1:35.40 /0.00 (952.81) @:0 B@:38 @0:0 @1:0 W:0
2018-07-25 16:51:56,519 - Recv: ok
```

For your tool heatup command however, the printer sends two temperature responses and then stops sending anything for at least 30s (within which your printer should have definitely sent quite some `echo:busy: processing` and temperature reports), after which OctoPrint gives up on it:

```
2018-07-25 16:51:56,577 - Send: N9 M109 S220 T0*37
2018-07-25 16:51:56,587 - Recv:  T:48.62 /220.00 (908.25) B:60.20 /60.00 (855.94) T0:48.62 /220.00 (908.25) T1:35.52 /220.00 (952.44) @:0 B@:31 @0:0 @1:0 W:?
2018-07-25 16:51:56,911 - Recv:  T:48.55 /220.00 (908.50) B:60.22 /60.00 (855.81) T0:48.55 /220.00 (908.50) T1:35.67 /220.00 (952.00) @:250 B@:27 @0:250 @1:250
2018-07-25 16:52:26,949 - No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
```

That also matches the information in `octoprint.log`:

```
2018-07-25 16:50:38,666 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2018-07-25 16:50:38,713 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-07-25 16:50:40,818 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, telling it to set the busy interval to our "communicationBusy" timeout - 1s = 4s
2018-07-25 16:52:01,922 - octoprint.util.comm - INFO - Timeout while in an active heatup, considering heatup to be over.
2018-07-25 16:52:26,948 - octoprint.util.comm - INFO - No response from printer after 6 consecutive communication timeouts, considering it dead.
2018-07-25 16:52:26,972 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
```

Based on this I would say it's some issue with the firmware of the printer, but you state that it works when you connect from a computer. So I got a couple questions from you:

  * What are you using on your computer to connect to your printer?
  * Is this a new install of OctoPrint or an updated one? If so from which version were you coming and was it working with that? 
  * Does this only happen when heating the hotend or does it also happen with the bed?
  * Does it always happen or is it intermittent?
  * When did you last update your printer's firmware?
  * Is this stock Marlin or some modified vendor version?

foosel, 2018-08-01:
Number 2 is still missing some answers ;)

Number 5 makes me wonder whether this means you updated the firmware and then the issue started arising (in which case a firmware bug sounds more likely) or whether you just updated the firmware by coincidence and the issue was happening before as well.

My goal here is to try to figure out when things started going wrong for the first time, what might have changed that triggered this "going wrong" and to get some information to allow to figure out what it might even be that's going wrong.

rashn, 2018-08-04:
I do not know what it was, a few days later the printer start works without problems. Maybe there was a problem with the power supply? In any case, the problem is not in your wonderful application. Thank you for your efforts. Issue closed.

closed 2018-08-04 by rashn

foosel, 2018-08-06:
Thank you for reporting back and closing the issue.