OctoPrint/OctoPrint #2759: After upgrade  octoprint is not starting

acocalypso, 2018-07-25:
#### What were you doing?

1. Installed update via webinterface 1.3.8

#### What did you expect to happen?
Octoprint will start without issues

#### What happened instead?
Infopage showed up that octoprint is not up
I checked the running processes and octoprint wasn't running
I saw in the octoprint.log that there was an error with PrintTImeGenue so I removed it with
`pip uninstall Octoprint-PrintTimeGeniue`

Plugin was successfully removed
I did a reboot of the Raspberry Pi3
Octoprint still not working

`2018-07-25 20:20:43,697 - octoprint.startup - INFO - ******************************************************************************
2018-07-25 20:20:43,700 - octoprint.startup - INFO - Starting OctoPrint 1.3.9
2018-07-25 20:20:43,701 - octoprint.startup - INFO - ******************************************************************************
2018-07-25 20:20:44,104 - octoprint.startup - INFO - Blacklist processing done
2018-07-25 20:20:44,194 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-25 20:20:47,160 - octoprint.plugin.core - INFO - Found 20 plugin(s) providing 20 mixin implementations, 19 hook handlers
2018-07-25 20:20:47,350 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-07-25 20:20:47,411 - octoprint.server - INFO - Intermediary server started
2018-07-25 20:20:47,412 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-07-25 20:20:47,967 - octoprint.plugin.core - INFO - Found 20 plugin(s) providing 20 mixin implementations, 19 hook handlers
2018-07-25 20:20:48,025 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 9, in <module>
    load_entry_point('OctoPrint==1.3.9', 'console_scripts', 'octoprint')()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/__init__.py", line 599, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/cli/server.py", line 188, in serve_command
    ignore_blacklist)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/cli/server.py", line 109, in run_server
    octoprint_server.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/server/__init__.py", line 284, in run
    analysisQueue = octoprint.filemanager.analysis.AnalysisQueue(analysis_queue_factories)
TypeError: __init__() takes exactly 1 argument (2 given)
`

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

RaspberryPI 3

I have read the FAQ.

khseal, 2018-07-25:
My Octoprint broken after updgrade.
Error log:
khseal@C110:~/OctoPrint/venv/bin$ ~/OctoPrint/venv/bin/octoprint serve
Traceback (most recent call last):
  File "/home/khseal/OctoPrint/venv/bin/octoprint", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/home/khseal/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/home/khseal/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/khseal/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: pkginfo>=1.4.2,<1.5

khseal, 2018-07-25:
Full reinstall octoprint.
Other errors:
(venv)khseal@C110:~/OctoPrint$ ~/OctoPrint/venv/bin/octoprint serve
Traceback (most recent call last):
  File "/home/khseal/OctoPrint/venv/bin/octoprint", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/home/khseal/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/home/khseal/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/khseal/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: monotonic>=1.3,<1.4

sticilface, 2018-07-25:
Me too. Total failed upgrade to 1.3.9 and failing to start. Tried reinstall. Will start from command line but service fails to load. Tried so many fixes now I’m just going to flash new sd card and start over!

khseal, 2018-07-26:
I'm start command on console ./venv/bin/python setup.py install
Octoprint started.

foosel, 2018-07-26:
@acocalypso I'll need some more info in order to be able to analyse this.

What version of OctoPrint where you coming from (I figure 1.3.8?)? You state you are running this on Raspberry Pi and the paths in your log suggest it's OctoPi - which version of OctoPi? What does `~/oprint/bin/pip --version` and `~/oprint/bin/python --version` return?

@sticilface same for you - more info please. "It doesn't work" is not something I can work with in order to help you. I need logs, error messages, versions etc.

sticilface, 2018-07-26:
I realise that my info was not particularly helpful.  unfortunately i cut my losses, backed the .octoprint folder up and flashed a new image!  needed to print something last night!  now running 1.3.9 no problems!

acocalypso, 2018-07-27:
@foosel:

I upgraded from 1.3.8 to 1.3.9

python: 2.7.3
pip: 9.0.1

octopi Image was 0.15.1

foosel, 2018-07-27:
@zero-llamas could be. I test against older versions of OctoPi/Raspbian, but not as old as that Python version seems to imply. 2.7.3 is fairly ancient. What output does `~/oprint/bin/pip install "python-dateutil>=2.6.1,<2.7"` produce?

@acocalypso, Something doesn't seem right here. OctoPi 0.15.1 comes with Python 2.7.13:

```
------------------------------------------------------------------------------
OctoPrint version : 1.3.9
OctoPi version    : 0.15.1
------------------------------------------------------------------------------

pi@octopi2:~ $ ~/oprint/bin/python --version
Python 2.7.13
```

I tested the upgrade path from 1.3.8 to 1.3.9 on OctoPi 0.14, 0.15.0 and 0.15.1 in various combinations. No issues here.

@acocalypso & @zero-llamas could you please share your `~/.octoprint/logs/plugin_softwareupdate_console.log` here? Download it through something like WinSCP, rename it to end in ".txt" and then just attach it to your comment.

zero-llamas, 2018-07-27:
@foosel output from pip command:
 ~/oprint/bin/pip install "python-dateutil>=2.6.1,<2.7"
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in ./oprint/lib/python2.7/site-packages
Requirement already satisfied: six>=1.5 in ./oprint/lib/python2.7/site-packages/six-1.10.0-py2.7.egg (from python-dateutil<2.7,>=2.6.1)
You are using pip version 9.0.1, however version 18.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

I think there may be a clue in those last couple of lines ;)
[plugin_softwareupdate_console.log.txt](https://github.com/foosel/OctoPrint/files/2235661/plugin_softwareupdate_console.log.txt)

I have attached the log file, but since I'm running OctoPi 0.12.0 I think I'm just dragging this issue in the wrong direction. I'm going to try https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/ and see how it goes.

Update: after upgrading using those instructions it has come back up again. So, in my case at least, it was simply old age. It might be worth doing some sanity checks on the version of python and pip before updates, and throw an error or warning if it's much outside your testing setup.

Thanks for your help. @foosel still loving Octoprint!

AndKe, 2018-07-27:
I should have not trusted a baked German :) 
"503 Service Unavailable" after restart when it did not came up.

- is there a  fix ?

foosel, 2018-07-27:
@AndKe I'm unable to comment unless your share some logs and version info. See my corresponding requests above.

AndKe, 2018-07-27:
did a apt-get update and upgrade..
I tried https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/

my status is now:
pi@octopi ~/OctoPrint $ ~/oprint/bin/pip install "python-dateutil>=2.6.1,<2.7"
Requirement already satisfied: python-dateutil<2.7,>=2.6.1 in /home/pi/oprint/lib/python2.7/site-packages/python_dateutil-2.6.1-py2.7.egg (2.6.1)
Requirement already satisfied: six>=1.5 in /home/pi/oprint/lib/python2.7/site-packages/six-1.10.0-py2.7.egg (from python-dateutil<2.7,>=2.6.1) (1.10.0)

now, on service restart I get the text 
"OctoPrint server started, waiting for it to get ready..." 
, and then the endless wait

 cat /home/pi/.octoprint/logs/octoprint.log |tail
2018-07-27 20:16:37,567 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.8 (Github)"
2018-07-27 20:16:37,823 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting

The two final log lines for OctoPrint are from before upgrading...

AndKe, 2018-07-27:
then, after a long wait, it suddenly is back ! :) 
Thank you.
hope you get reasonable temperatures soon

foosel, 2018-07-27:
Thanks, so far it's not looking like it'll cool down soon.

Honest question for both you @AndKe and @zero-llamas - I take it you hadn't yet followed the announcement before? If so, first of all, why? It's not like I put that out just for the fun of it ;) And second of all: past updates through OctoPrint worked anyway? That puzzles me greatly.

AndKe, 2018-07-27:
@foosel 
>>"I take it you hadn't yet followed the announcement before?"
I do not know what exact announcement you mean.
I am an eager reader of releasenotes, - reading about the cancel/pause fix made me upgrade ASAP (that was annoying, so thanks for fixing)
Few days back I read about the security warning, and checked the version, regardless of the printer being inaccessible from outside without VPN.

So I have the OctoPi 0.12.0 , now updated. it's not exposed to the internet, so I do some updating to the OctoPi .. yearly .. or less.
It worked all the years, except once , you helped me out with https://github.com/foosel/OctoPrint/issues/2193
- then it worked.. every time till now.

I do not honestly know what announcement you mean but it's not that easy to know what OctoPi version/update status one is at (so a quick mention may be overlooked ?), maybe you need to check for some dependencies, and give us a kick in the ass by denying upgrade if you see too old... pip (?)

zero-llamas, 2018-07-27:
@foosel yep, similar story. I tended to hit upgrade every time it told me there was a new version of octoprint.  I must have missed the original message about upgrading python, the only reason I found it was from googling "upgrading python on octopi" when I realised it was all too old.

foosel, 2018-07-28:
You both should have gotten a big red important looking notification about the link you both now followed to fix things, back in November: 

[image]

Unless you disabled the announcement plugin which you really should for these exact kind of reasons since it's the only way to reach people with important notifications like "yo, do this or things will stop working" who don't follow OctoPrint on social media or directly on the blog.

> maybe you need to check for some dependencies, and give us a kick in the ass by denying upgrade if you see too old... pip

Sorry, but I can't anticipate every single potential incompatibility under the sun and prepare for it while also doing *everything else I already do*. In this particular case even less though due to the nature of the issue. I blast out warnings for such stuff *through OctoPrint* as soon as I notice them. Following the links explains exactly how to check if one is affected and what to do to fix things.

I run this project mostly alone, take care of the ongoing maintenance, future updates, compatibility tests as far as I can justify them, have to look into funding, manage the community, provide support, try not to grow too frustrated at some of the very infuriating messages I receive, get screamed at for updates gone wrong due to reasons outside of my control, am expected to work around broken firmware, solve issues before they even arise and implement new functionality ideally yesterday because it's so important for someone's workflow right now.

I do all of that as far as humanly possible. Please do your part and follow warnings I put out *when I put them out* and not 9 months later ;)

foosel, 2018-07-28:
Sorry, that was poorly phrased on my side - I didn't mean that you blamed me, but on every update I get people that are... let's just say "unfair" ;) and who throw around a lot of blame and don't even consider the issue maybe residing on their end. It was those people I was referring to with regards to all the stuff I already handle which simply doesn't leave a lot of room for further anticipation and resilience against possible sources of errors during operation.

I agree that the notifications in general need a different handling, that's something I've been thinking about how best to solve too.

MechaNickW, 2018-09-01:
I am having a similar issue. I am running an OctoPi 0.12.0 on a Raspi model B, upgrading from 1.3.8 to 1.3.9. I did apply the fixes in the bright red notification way back when they first came out, have kept current on updates ever since with no issue. I get the 503 Service Unavailable error when trying to log into Octoprint, and have tried manually restarting with no success through SSH. I'm trying to reinstall Octoprint manually from the command line, if that doesn't work I will upgrade my OctoPi image. I'm not complaining or expecting a fix on this ancient of a setup, just wanted to identify that this did occur on an instance with the applied pip/python patches.

Gina, thank you SO MUCH for your continued work on this project. I support through Patreon, it's worth every cent and I hope you continue work on Octoprint - it's literally my favorite piece of the 3D printing software toolchain.

foosel, 2018-10-29:
Unless I'm mistaken (if so, please speak up), this is fixed by the steps outlined [in this post](), so I'll close this.

I don't understand why that fix apparently didn't take hold for some but I honestly can't justify spending more time looking into this considering how old 0.12 is at this point - sorry. 1.3.10 will ship with a bundled backup plugin that allows you to easily migrate to fresher images though.

closed 2018-10-29 by foosel