OctoPrint/OctoPrint #2760: Octoprint reporting wrong version to Slic3r, which keeps it from connecting.

MStylesMS, 2018-07-26:
#### What were you doing?

Trying to connect to OctoPrint from Slic3r 1.40.1+.

1. Go to Printer Settings tab.
2. Input correct hostname, API key, etc.
3. Click Test.

#### What did you expect to happen?

Successful connection from Slic3r.

#### What happened instead?

Got an error in Slic3r saying "...Your browser sent an invalid request.  Note: OctoPrint version at least 1.1.0 is required."

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.3.9 general release.

#### Operating System running OctoPrint

Raspian

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Prusa MK2

#### Browser and version of browser, operating system running browser

Slic3r's internal browser.

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-07-26:
I just ran a `GET` request against an OctoPrint 1.3.9 on `/api/version` and am getting back what I'd expect:

```
{
  "api": "0.1", 
  "server": "1.3.9"
}
```

Am I right in assuming that the Slic3r version you are referring to here is in fact not the mainline Slic3r but instead Slic3r PE? 

I tried with an install of Slic3r PE 1.38.5 I still have lying around and get no error:

[image]

I then downloaded Slic3r PE 1.40.1. Also no error:

[image]

And while I was at it I also downloaded Slic3r PE 1.41.0-alpha2:

[image]

The "Browse" button also properly shows the versions of all the instances here, some of which are 1.3.9, some of which are 1.3.10.dev:

[image]

So... I cannot reproduce this. Any more hints?

MStylesMS, 2018-07-26:
Thanks for the quick attention to this!

I rebooted everything and still get the same error.

I then browsed for it on the network (again…I had done it before) and selected it again tried Test.  This time I got a slightly different error that said I needed a newer version BUT it also complained about an invalid API key.

I checked the API key visually and it looked the same.  I had copied and pasted it and the first and last characters were the same.  However, copied it again and this time it worked as expected.

For reference, it will connect both by local IP address as well as an OpenDNS connection from outside my network.

It appears that the problem may have been that it had an incorrect API key, but that puzzles me because I had copied, pasted, and visually checked it before.  Possibly the reboot affected things?

Regardless, the prior error message I had received did not mention the API key.

I will close the ticket and keep watch for it to happen again.  I will take better notes and screenshots then.

Mark Stevens
Co-Founder & Vice-President

closed 2018-07-26 by MStylesMS

foosel, 2018-07-26:
> It appears that the problem may have been that it had an incorrect API key, but that puzzles me because I had copied, pasted, and visually checked it before.  Possibly the reboot affected things?

That would be a first one. I have no clue really.

> Regardless, the prior error message I had received did not mention the API key.

That's probably something best taken up with either the Prusa Slic3r devs or mainline Slic3r - I'm not responsible for the accuracy of their error messages ;)

CostaRich, 2019-05-07:
I think I know what happened, because this same thing happened to me. I'd given up and was just dropping the gcode files on the html home page to print.  I thought I just had a version mismatch and it would get fixed whenever I upgraded firmware.  So I just did that on both the printer and the octoprint pi box, and still, no joy.  

So  as Mark did, I also checked the first and last characters of the API key and it looked OK, but wouldn't work (same message). I eventually tried using the default original Prusa profile and it did work.  So I saved a copy of the original .INI file and diff'd them and the issue is obvious when you look at the .INI file but is not at all obvious from the UI.  The UI somehow lets you have multi-line input or something when you paste in the API.  If you aren't careful, you can end up with two API keys separated by a \r, which gives you two 'lines' of input in that one teeny text box.  

The solution would seem to be having that API box be large enough that you can actually see the entire
key at once (and then some) so you'd know you had a data error without having to fiddle around in that text box.  Also would be better for Slic3r to report a better error as it is misleading.   That's something for Prusa to figure out.  

If you want to reproduce it, it is very easy. 

Here's what my .INI file looked like.  (Neither of these API keys are live anymore)

printhost_apikey = AEF0165AA401077BB447A55489F72A62\r4B895550008E40FDB83FE2DDFD318490

So I'm sure what happened is I hit 'paste' at some point but didn't have the previous key selected.  But when you look, it's hard to tell you now have two API keys in one box.  Worse, it's hard to fix.  I double-clicked and pasted in the new API key several times and it didn't solve the problem because double-clicking the API key only selected the first one.  If you change your .INI file and load the key above you'll see what I mean, the box acts squirrelly.  You can hit the up and down arrow and flip between the two keys.  And if you hit Command A to select all, that only selects the first API key.  That \r causes serious mayhem.  I'm guessing it also causes the server grief as the return code is misleading in that it is broken and says (literally, those html codes are printed as literal values in the dialog box)

Could not connect to OctoPrint: HTTP 400: <html><body><h1>400 Bad Request</h1>
Your browser sent an invalid request.
</body></html>

Note: OctoPrint version at least 1.1.0 is required.

Anyway, this is what I found the problem to be after looking at it for a few hours.

Cheers,

Rich

CostaRich, 2019-05-07:
BTW, note that even though I typed out the HTML codes as they were displayed in the message, the GitHub forum software nicely interpreted it as markup code. Below is a screen shot of it.

[image]

XOIIO, 2019-06-19:
Any updates on this? Just ran into this issue myself which kind of puts a crimp in the ease of use of octoprint.