OctoPrint/OctoPrint #2763: MinTemp error triggering print stop.

Cr0nJ0b, 2018-07-27:
I've just updated to the latest 1.3.9 octoprint.  The update process was pretty onerous, but I got through it.  I could never never get polarCloud to properly update, so i just removed the plug-in.  

I'm running an ANET A8 with the latest marlin firmware.

My issue is that I'm now getting random MinTemp errors that are stopping the print jobs.  They always occur on the build plate ID.  

I'm watching the temps and they are relatively stable, but I don't know what the tolerance is set for.  It fluxes between 55 and 54 degrees for a 55 setting.

Some prints complete fine and others die in 10 minutes.

Is this a known issue?  Can I just keep Octoprint from shutting down the printer?  I get that the firmware might be doing this, which is where I'm going next...but wanted to see if this was a known octoprint issue.

thanks

foosel, 2018-07-27:
This is not a known issue and also nothing OctoPrint can be at fault for - mintemp errors are something exclusively triggered by your firmware.

Cr0nJ0b, 2018-07-29:
Thanks for the help.  Was a frustrating few days.  Turns out my temp wire was broken somewhere along the path.  This happened at the exact same time I did software updates.  8 hours later printer is back together and working with the latest Marlin.  I still can't get Octoprint to flash the printer directly, I've tried a ton of different things even up to AVRDUDE flash manually from the pi, but can't seem to get things working.  A minor inconvenience really.

have a great weekend.  Please close this thread.

closed 2018-07-31 by foosel