OctoPrint/OctoPrint #2766: No Extruder Switch on Prusa MK2 MultiMaterial (MM)

Flappie, 2018-07-28:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-08-20:
Your `serial.log` wasn't enabled (as it also told you in the `serial.log` itself). Please enable it, reproduce the issue *with the provided file* (and make sure you test *if it even reproduces in safe mode at all!*) then post the resulting `serial.log`.

As a side note: OctoPrint runs prints by sending the lines from the printed file as-is to the printer (caveat: they might be modified by third party plugins you have installed). It's the responsibility of your printer to do the right thing based on the commands it receives. Looking into the file the printer should receive these commands:

```
```

If OctoPrint sent those but the printer still misbehaved, something's up with the printer. If OctoPrint didn't send those we will need to find out why, so check if a) the issue even persists in safe mode (it might a third party plugin messing with things) and if so b) enable & create a `serial.log` of the issue so it's possible to compare what should have gone over the line (file contents) vs what actually went over the line (sent/received commands according to `serial.log`).

Additional side note:

```
2018-07-28 19:56:23,957 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin V1.0.2; Sprinter/grbl mashup for gen6"
```

That is a Prusa M2S MMU? Odd, I thought the Prusa firmwares did report as "Prusa Firmware vx.y.z based on Marlin" or something..,

Flappie, 2018-08-24:
Hi Gina,

Thank your for your reply.
Sorry that i did not post the full serial.log.
At the time that i did the upgrade and after i did not have it enabled.
Doing a quick downgrade made the problem go away without chaning anything
else.

My main concern was that there was a logging of this event on github so
others that maybe also have this issue can relate and we can solve it
together.
As many people my time if very limited and i do not have much time to
reproduce it with the proper logging and safe mode on.
I am happy to reproduce it with the logs and safe mode but i don't have the
time (have to think about my wife and kid).
If i do find time i will try of course ;-)

Why my logging shows "Marlin xxx" is weird beause i have the Prusa 3.1.0
version on my printer....
Thank you for noticing.

Perhaps we suspend this topic for anybody else to find and if they do we
have more reason to investigate.
This saves you time and me.
If it does happen with anybody else i can make it a higher priority and
give a better (proper?) report.
Because it can be something on my side being plugins of something else(?).

Thank you for all your hard work and i'll keep donating to you on patreon.

Kind regards,

Erik Legerstee

Plastikschmiede, 2018-10-14:
Thank you @Flappie !

Thats the exact same Problem i have with the MK3 MMU2.0 and Octoprint   1.3.9

I hope the Problem can be fixed soon!

foosel, 2018-10-15:
> I hope the Problem can be fixed soon!

Not without the required logs, testing whether the issue persists in safe mode, sample files, ... In short - collaboration by people who have the hardware and overall setup to actually reproduce this issue. Which I'm still not even sure that it's even OctoPrint since as I already said, OctoPrint is basically a GCODE sender, and it's the job of your printer's firmware (and also your slicer in this particular case) to perform tool switches.

Flappie, 2018-10-15:
Hi Gina,

We (or I at least) don't expect anything from you at this time.
I think i'll wait for more people to show up with this same problem then i'm willing to make time.
At this time all i running fine with the current version.
I'm also about to upgrade my printer with new hardware and firmware then i'll check again.
@Plastikschmiede please feel free to produce the needed logfiles

Thanks again for the work @foosel 

Kind regards,

Erik

foosel, 2018-10-19:
Closing this then for the time being as in its current form it's not actionable. Happy to reopen once there are logs available which show an issue on OctoPrint's side :)

closed 2018-10-19 by foosel

fasteddy516, 2018-11-09:
I have the same issue - or at least the same symptom - and have had it for quite awhile; since February/March of this year, and through many versions of OctoPrint/Slic3r.  I haven't had time to do anything about it until now, so I've been doing all of my multi material prints straight off of the SD card, and using OctoPrint any time I print in single material mode.  

I finally got around to doing some testing with OctoPrint in safe mode using a gcode file that just contains the quad-material priming sequence.  Here is a link to a gist with octoprint.log, serial.log and the gcode file I used (logs aren't trimmed at all, so not all is necessarily relevant):

[https://gist.github.com/fasteddy516/dbc0bdc38e859ec8a67414bec6a5f058](url)

A quick look at the serial.log seems to show OctoPrint sending everything correctly, but the printer responds with "Invalid T Code".  The order of the serial TXs/RXs is a bit wonky in places, but I'm not sure if that is significant.  It certainly looks like this is an issue on the printer side rather than OctoPrint, but I figured I would add my "me too" here, and share the logs.  I've also got the hardware (Prusa MK2 MMU) and potentially some time to do additional testing if more information is needed.