OctoPrint/OctoPrint #2768: [improvement] h.264 support?

Harryb1991, 2018-07-29:
It seems ffmpeg is outdated these days and too power hungry for any RPi, I spent £100 on a decent webcam (my own stupid fault) because I presumed octoprint would support h.264 and in this day and age, HD is the least you expect. I almost cried when I got 10fps at only 720p 👎 . Im not sure if its possible but it makes sense, surely?

This is literally the only issue I have with octoprint, other than that, you do an amazing job @foosel. Thanks for your work mate.

Harryb1991, 2018-07-29:
Also, while Im here, is there any way of running a script to trigger a control in octoprint. i.e a script to stop, start, pause?

Thanks in advance guys 👍

ntoff, 2018-07-29:
>cried when I got 10fps at only 720p

OctoPrint itself doesn't control the webcam, that's the job of mjpeg streamer, which you can change by editing /boot/octopi.txt, the directions for doing so are in the FAQ https://discourse.octoprint.org/t/how-can-i-change-mjpg-streamer-parameters-on-octopi/203 which you should probably go through.

Octo**Print** (the issue tracker you're currently posting in) also doesn't include *any* webcam streaming software, you're probably thinking of [Octo**Pi**](https://github.com/guysoft/OctoPi/) which bundles mjpeg streamer along with some other goodies. With OctoPrint, you're free to install whichever webcam streaming software you like that is compatible with your system.

Also, your request isn't the first of its kind, *many* have already requested this feature.

>Also, while Im here, is there any way of running a script to trigger a control in octoprint. i.e a script to stop, start, pause?

For help and support, please use the forum located at https://discourse.octoprint.org/. Thankyou.

Harryb1991, 2018-07-29:
Hi, thanks for taking the time to reply to me. I didn't know that the two were different, I've no experience with Octoprint on any platform other than a RPi. I wrote ffmpeg, I was meaning mjpeg-streamer. I'm aware that mjpeg-streamer handles compression and is whats used in the web server.. Using UV4L for h.264 would be so much smoother and would probably help with latency issues Im having whilst away from home. It seems a no brainer to me. I know I can set up a stream seperate to octoprint with UV4L but it would be a lot nicer if it was embedded into Octopi.

I will take that the answer to my request is no then. Damn, I wanted to hook a Google AIY kit to Octoprint.