OctoPrint/OctoPrint #2770: error: [Errno 97] Address family not supported by protocol

radical-equanimity, 2018-07-29:
I need to add "HAS_IPV6 = False" in .local/lib/python2.7/site-packages/octoprint/util/net.py for the server to come up. I have kernel module ipv6 blacklisted.

ntoff, 2018-08-01:
Please fill out the ticket in its entirety, you can find all the information you need to include here https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report 

Did this happen with a previous version of OctoPrint?

radical-equanimity, 2018-08-01:
Issue was not present in 1.3.8.

I see that CHANGELOG.md has a section added last Wednesday regarding this
issue. "Disabled IPv6 and now there are issues with the server after the
update?"

OK to close issue I suppose, but solution of having user need to add Flags
to command in order to get around new regression is not elegant IMO.

https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report

foosel, 2018-08-01:
It's 2018. I'm not entirely sure why someone would even disable IPv6 considering that IPv4 is pretty much end of life, and this note was added to the release notes for the final release that happened last week for the few cases that might know their way around computers enough to even know what IPv6 is and how to disable it that this small change shouldn't pose a real problem, after I saw some drive by message from someone (you?) who'd run into some problem but was gone so fast it was impossible to follow up with details.

OctoPrint checks whether Python's socket module is reporting IPv6 support and only if it does will attempt binding to V6 addresses. Apparently in your case that socket module will claim there's IPv6 support on the system even though it isn't. Which I'd rather consider a bug in Python's socket module than an issue with Python code depending on information provided by the underlying system.

closed 2018-08-01 by foosel

radical-equanimity, 2018-08-01:
Gina, users may need to disable IPv6 for various reasons. It should not be
assumed otherwise.

I agree that the bug appears to exist not in your code, but rather in the
Python socket module. You took the precaution of checking for support
rather than forcing use of IPv6 and also offer a workaround. This allows
users with similar setup to continue operating as usual.

hemna, 2018-10-13:
Believe it or not, but the vast majority of us never use ipv6, especially for home networks.  It's completely unnecessary.

damianharouff, 2018-10-23:
What a patently horrible response from the maintainer. Python 2.7 is "pretty much end of life" but yet it's still in use with Octoprint? Ridiculous.

My specific scenario is that printers aren't set up with IPv6 since they're on a wireless bridge that doesn't pass IPv6 over it. I'm not going to buy new hardware for no other reason than to have IPv6 on a local network.

With failure to implement proper error control beyond a response of "It's 2018", makes me wonder where other sections of Octoprint have no/poor error control, including safety features. Looks like it's time to look elsewhere for printer control software.

sanchosk, 2018-10-28:
> I need to hardcode "HAS_IPV6 = False" in .local/lib/python2.7/site-packages/octoprint/util/net.py for the server to come up. I have kernel module ipv6 blacklisted. Error occurs line 1552 of .local/lib/python2.7/site-packages/octoprint/server/__init__.py

Thanks for the hint, I'll try this and perhaps I'll be now able to continue using octoprint...
Just a small correction - the line is
HAS_V6 = False
not IPV6

jubaleth, 2018-10-29:
Morning!

So, firstly, the response from Gina was neither 'horrible' nor was it 'arrogant', it was simply a statement of fact.

> Python 2.7 is "pretty much end of life" but yet it's still in use with OctoPrint? Ridiculous.
I invite you to take a look at the PRs currently open, and to take note of all the work being done to change this. I can assure you, upgrading a language by a major version is no small feat, especially when you need to maintain backwards compatibility. 

Another point, is that this issue has *nothing* to do with OctoPrint, this has to do with a python module incorrectly claiming that ipv6 is available. I get your frustration, but this is *really* the wrong place to express it. 

> I'm not going to buy new hardware for no other reason than to have IPv6 on a local network.
This is exactly the mentality that is slowing down the rollover to IPv6. The fact of the matter is, it *is* 2018, and more and more ISPs are allocating IPv6 blocks to home networks. Just because *you* don't feel like upgrading, doesn't mean it isn't happening.

Just to note, the release notes do specify a configuration that allow you to direct OctoPrint to only bind to v4 addresses (i.e. host="0.0.0.0").

I'm locking this ticket now, before this continues down the road of being completely unproductive.