OctoPrint/OctoPrint #2772: Error detection works at 30% or less

AndKe, 2018-07-30:
#### What were you doing?

I described before, in another issue, that I mean to have observed missed failure detection. This time I reproduced it at will.

The next time, (no disconnection/reconnection, just M999 and retry , the OctoPrint failed to detect any "Error" and continued to dump data to the printer as if nothing happened.)

### Steps to Reproduce

While printing , provoke a thermal runaway (turn off power til heater(s))

#### What did you expect to happen?

I'd expect OctoPrint report the first error: (Thermal runaway)

#### What happened instead?

The first error did not get recognized, instead, OctoPrint got the secondary error which is less verbose, and does not tell the actual issue.

[image]

#### Did the same happen when running OctoPrint in safe mode?

not relevant
#### Version of OctoPrint
1.3.9

#### Operating System running OctoPrint
OctoPi 

#### Printer model & used firmware incl. version

homemade , Marlin 1.1.8

#### Browser and version of browser, operating system running browser

Linux, Chromium

#### Link to octoprint.log
[octoprint (2).log](https://github.com/foosel/OctoPrint/files/2240594/octoprint.2.log)

#### Link to contents of terminal tab or serial.log
[serial.log](https://github.com/foosel/OctoPrint/files/2240598/serial.log)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
above
[image]

foosel, 2018-09-29:
To quote #2764 

>Kill situations now have a special handling thanks to a couple of commits pushed on `maintenance`, soon `devel` and to be released with 1.3.10.
>
>See https://github.com/foosel/OctoPrint/compare/87e7a19fdda3...f88147003c0c for details.

foosel, 2018-12-10:
1.3.10 has been released.

closed 2018-12-10 by foosel