OctoPrint/OctoPrint #2773: Filament amount not being calculated correctly.

BeTen, 2018-07-30:
#### What were you doing?

1. Startet Octoprint on an Raspberry Pi
2. Uploaded the gcode for my print (for reference i added it here)
[72_Wandadapter_3x.zip](https://github.com/foosel/OctoPrint/files/2242304/72_Wandadapter_3x.zip)
3. It shows me the following output:
[image]

#### What did you expect to happen?
Showing the correct amount (or estimated amount) in cm³.
#### What happened instead?
Well, you can see it in the screenshot, this would be 1077 Liters of filament, which, yeah, would need a truck to bring the spool. 
Also the telegram plugin sends the same wrong value, as well as the cost calculator.
#### Did the same happen when running OctoPrint in safe mode?
Yep. The screenshot was taken already in safe-mode.

#### Version of OctoPrint
OctoPrint 1.3.9 auf OctoPi 0.13.0

#### Operating System running OctoPrint
OctoPi 0.13.0

#### Printer model & used firmware incl. version
Anet A8 with Skynet3d.

#### Browser and version of browser, operating system running browser
Google Chrome Version 68.0.3440.75 (offcicial build) (64-Bit)
OS: Win7 64Bit

#### Link to octoprint.log
https://gist.github.com/BeTen/95bcd05ae9ba07d5cf4b6b8a745facff

#### Link to contents of terminal tab or serial.log
Well, communication to the printer does not be needed to reproduce the problem.

#### Link to contents of Javascript console in the browser
-

#### Screenshot(s)/video(s) showing the problem:
Screenshot see above

I have read the FAQ.

foosel, 2018-08-20:
I cannot reproduce this with the provided file (renamed in the screenshot), freshly uploaded on a 1.3.9 instance via the web interface:

[image]

Was this actually analysed under 1.3.9 or might it have been analysed before? Was it uploaded through the web interface or maybe through slic3r?

BeTen, 2018-08-20:
Hi,

will check and try to reproduce again when I‘m back from vacation. Thanks
so far!

Best regards,

foosel, 2018-08-20:
>  when I‘m back from vacation

Rough time frame when that will be, just so I know when I should start to worry if this got forgotten? ;)

BeTen, 2018-08-20:
Leaving greece on friday, will arrive back on monday, try to check next
week - or if my VPN works earlier :)

foosel, 2018-08-20:
Perfect, thanks! And have a nice vacation!

truglodite, 2018-09-02:
I had the exact same behavior with filament volume calcs while I was shifting around through various plugins. The problem 'magically' went away, but I made so many changes between that I can't say what fixed it exactly. Among other possible things, I removed printer stats, updated raspbian, re-entered 1.75mm diameter filament manager, and added many other plugins including print time genius. I still had to go back and enter corrected volume values in print history, but since it fixed itself it hasn't acted up again.

BeTen, 2018-09-03:
Hi,

sorry for the late update - had issues with the printer - but today, I
still had the same behaviour.
Files were uploded through the Webinterface.
All components are up to date (did a „search for updates“ -> Everything is
up to date).
The Slicer being used was „Slic3r Prusa Edition“.

Best regards,

cicinovec, 2019-08-31:
I have the same issue today with the files attached.
[holder.zip](https://github.com/foosel/OctoPrint/files/3562133/holder.zip)

When slicing in cura, I get 1:55 hours. Then uploaded to Octoprint the time is 176 hours with tons of filament.

[image]

Octoprint and all plugins are up to date. Tried with defaul octoprint analyser and also with Print time genius, same result

I have "fixed" it by using old version of Cura, but since this is the only model causing the issue, I dont believe it is only Cura issue.

BeTen, 2019-08-31:
Have you changed plugins recently?

cicinovec, 2019-08-31:
No, the installation is the same, no changes except some updates. Also no new plugins installed

cicinovec, 2019-09-01:
Today test:

Cura 4.2.1 - issue with calculation with both calculators: default Octoprint calculator and Print time genius
Cura 4.0 - wrong calculation only with default calculator, after recalculating with Print time genius the time and filament amount is correct.
Model and the last log in the zip file
[InsertSite2.zip](https://github.com/foosel/OctoPrint/files/3563175/InsertSite2.zip)

foosel, 2019-09-04:
So... your file contains absolute coordinates, however right at the beginning of `Holder.gcode` there's a line

```
G91 F1500;
```

A `G91` means relative coordinates, which explains the ridiculous model size and travel lengths that are calculated. The F-parameter makes no sense here, which makes me thing that should have been another G command maybe, e.g. just a G1 for setting initial feedrate to 1500? In any case, the analysis is correct based on the commands therein, even though said commands don't make much sense in this combination. Removing the wacky G91 yields reasonable results:

[image]

Putting `InsertSite2.gcode` through the analyser doesn't produce a result on stock that looks unreasonable, but I lack the STL or further data to confirm or deny. If you think something's off with that please open a *new ticket*.