OctoPrint/OctoPrint #2774: [BUG] Autoscroll stops autoscrolling on switching browser Window (Chrome MacOS)

bolsoncerrado, 2018-07-31:
#### What were you doing?

Using Octoprint, Switching from TERMINAL tab to other tabs/browser windows.

#### What did you expect to happen?

AutoScroll on Terminal window to actually autoscroll.

#### What happened instead?

Instead, it stopped autoscrolling. It only happens when switching WINDOWS on the browser (ie going to google on a new TAB for example). It does NOT happen if I switch Octoprint tabs within the same octoprint browser tab.

#### Did the same happen when running OctoPrint in safe mode?

Didnt try.

#### Version of OctoPrint

1.3.9 but was also present on 1.3.8 at least.

#### Operating System running OctoPrint

Raspbian Jesse

#### Printer model & used firmware incl. version

Prusa MK3

#### Browser and version of browser, operating system running browser

 called from MacOS HighSierra Chrome browser 67.0

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

ntoff, 2018-08-01:
Seems like a chrome on osx issue, another user reported it https://github.com/foosel/OctoPrint/issues/2716 but failed to really give any details, although they did say it works in firefox (have you tried firefox to see if your results are the same?).

foosel, 2018-08-20:
@bolsoncerrado could you please report back whether it works in firefox for you?

I'm also unable to reproduce this against Chrome 67 on High Sierra myself. Opened OctoPrint, switched to the terminal tab, observed Autoscroll working. Opened a new tab, did a google search in it, scrolled around a bit, clicked through some pages, returned to OctoPrint tab. Observed that Autoscroll was still working as expected.

bolsoncerrado, 2018-08-20:
I just tried in Chrome 68 High Sierra and its definetly STILL happening.

Open a tab, load octoprint, click terminal during a print, scrolling great.
Open a new tab, load google
Return to octoprint tab, scrolling is lost.

This simple.

Does not happen with Firefox nor Safari :(

foosel, 2018-08-20:
[image]

This is High Sierra and Chrome 68 (running on Browserstack because I don't own a Mac). It's working as expected. Are you doing anything differently?

bolsoncerrado, 2018-08-20:
NO. Exactly that renders the autoscroll thing issue. How did you record that, so I can show you?

foosel, 2018-08-20:
I used a software called [licecap](https://www.cockos.com/licecap/).

I also just took a look at the code in question and the only possible way I see that autoscroll could disable itself is if a scroll event on the terminal was triggered. That didn't happen though when it autoscrolled for me. Might be some odd race condition though, possibly. In that case it would be very odd though that it only reproduces on Chrome on MacOSX.

When it happens, does the "Autoscoll" button pop back out:

[image]

Or does it still claim to be pressed:

[image]

?

Can you also *please* test if the issue persists in safe mode? You are prompted to do that in the issue template for a reason.

bolsoncerrado, 2018-08-20:
They're all printing now. Will try in safe mode next time, in the meantime check this out:

[image]

bolsoncerrado, 2018-08-20:
After the NOW LOOK AT THIS! it's STILL working but once again once I click the other tab, autoscroll stops working again. VERY weird!

foosel, 2018-08-24:
Good news: I can now reproduce this. On my laptop, running a Windows build of Chrome, so it's not restricted to MacOSX. And the cause is the exact location in code that I suspected, so it is in fact a race condition. 

The problem is that apparently on browser tab change a scroll event might get triggered (i have not yet found out WHY), at a point where the terminal isn't scrolled fully to the end (since new data arrived), and that will then get interpreted as a scroll by the user, disabling auto scroll. That was a feature added by @kantlivelong in a7a78c11d4ab16a8bd6601350b94529aca01da0e and actually makes sense since you don't want to get scrolled back to the end on every received line if you are actively trying to scroll up.

I'm currently checking if it's possible to somehow prevent this code section from running if the scroll event doesn't stem from the user, or the scroll amount is too small, but so far my ideas haven't worked.

kantlivelong, 2018-08-24:
Ah interesting! Just a complete guess but perhaps checking `document.hidden` before `self.autoscrollEnabled(false);`?

foosel, 2018-08-24:
Hm... that might be an idea. Although I'm not 100% sure it's still hidden when this issue arises, but it's worth a try.

ntoff, 2018-08-25:
>That was a feature added by @kantlivelong

So does that make this plugin redundant now? https://plugins.octoprint.org/plugins/autoscroll/

kantlivelong, 2018-08-25:
I didn't even realize that existed. :man_facepalming:

foosel, 2018-08-27:
Regardless of plugins or not, I think I might have found a solution. At least I can no longer reproduce it with the above commit applied. Pushed to `maintenance`, soon `devel`, will be part of the 1.3.10 maintenance release.

foosel, 2018-12-10:
1.3.10 has been released.

closed 2018-12-10 by foosel