OctoPrint/OctoPrint #2777: skipped steps upon reload of website

universam1, 2018-08-02:
#### What were you doing?

When I reload the website of OP this reproducible causes steps or layers to be skipped. The same happens when I open the website from another device like a smartphone. When the page has been loaded before, the print is fine.
After upgrading of OP this issue started to raise.
Happens probably since version >1.3.5, I can reproduce the issue with 1.3.8 and update on 1.3.9 does not solve it.
The issue has not been on OP versions older than 1.3.6 as far as I remember, though hard for me to downgrade for tests.
Tried different Marlin FW versions for doubt, but has no influence. Tested with 1.1.8, bugfix and 1.1.9
Printing for years on this setup, Prusa i3 with OP running on OrangePi w/ 1GB RAM with CPU usage of <20% and RAM is mostly free

#### What did you expect to happen?
Reload of webpage should not cause print to skip steps
#### What happened instead?
skips steps
#### Did the same happen when running OctoPrint in safe mode?
uninstalled all plugins

#### Version of OctoPrint

1.3.9 and 1.3.8

#### Operating System running OctoPrint
Armbian on OrangePi
CPU load is under 20% all the time

No Serial errors logged during the failure

#### Printer model & used firmware incl. version
Prusa i3
Marlin FW 1.1.9
#### Browser and version of browser, operating system running browser
Chrome on W10
Chrome on IOS
#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

ntoff, 2018-08-02:
https://github.com/foosel/OctoPrint/issues/1241 Probably a duplicate of this

foosel, 2018-08-20:
I agree with @ntoff, sounds like a duplicate.

closed 2018-08-20 by foosel