OctoPrint/OctoPrint #2778: Server Internal error

gege2b, 2018-08-03:
hi there

#### What were you doing?

I was messing with upgrades (I had the sarge 0.1.5 bug so I downgraded it to 0.1.4)
Now I'm running OP 1.3.9 and I get an internal error, looking in the log there is something about flask (see above)

#### What did you expect to happen?
being able to access the UI

#### What happened instead?
Error 500

#### Did the same happen when running OctoPrint in safe mode?

Yes, same problem

#### Version of OctoPrint

1.3.9 latest

#### Operating System running OctoPrint

raspbian stable

#### Printer model & used firmware incl. version

n/a

#### Browser and version of browser, operating system running browser

firefox 61 on win10

#### Link to octoprint.log

```
2018-08-03 16:01:17,002 - octoprint.startup - INFO - ****************************************************************                                                                                                                        **************
2018-08-03 16:01:17,004 - octoprint.startup - INFO - Starting OctoPrint 1.3.9
2018-08-03 16:01:17,005 - octoprint.startup - INFO - ****************************************************************                                                                                                                        **************
2018-08-03 16:01:17,070 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2018-08-03 16:01:17,075 - octoprint.startup - INFO - Blacklist processing done
2018-08-03 16:01:17,114 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/s                                                                                                                        ite-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-08-03 16:01:18,402 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-08-03 16:01:19,611 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't b                                                                                                                        e available
2018-08-03 16:01:19,629 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 20 ho                                                                                                                        ok handlers
2018-08-03 16:01:19,740 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-08-03 16:01:19,769 - octoprint.server - INFO - Intermediary server started
2018-08-03 16:01:19,770 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/s                                                                                                                        ite-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-08-03 16:01:19,780 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-08-03 16:01:21,134 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 20 ho                                                                                                                        ok handlers
2018-08-03 16:01:21,218 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octopri                                                                                                                        nt/uploads...
2018-08-03 16:01:22,971 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads in                                                                                                                        itialized successfully.
2018-08-03 16:01:23,174 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-08-03 16:01:24,973 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python2 -m pip" as command to invoke pip
2018-08-03 16:01:26,761 - octoprint.util.pip - INFO - Version of pip is 18.0
2018-08-03 16:01:26,763 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-08-03 16:01:26,763 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-08-03 16:01:26,770 - octoprint.plugin.core - INFO - Initialized 16 plugin implementation(s)
2018-08-03 16:01:26,783 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/discovery
|  FileManager (0.1.2) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filemanager
|  Firmware Updater (1.0.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  GcodeEditor (0.2.6) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_GcodeEditor
|  Logging (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/logging
|  Navbar Temperature Plugin (0.9) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_navbartemp
|  Octolapse (0.3.2) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_octolapse
[... 94 lines cut ...]
    rv = _json.dumps(obj, **kwargs)
  File "/usr/lib/python2.7/json/__init__.py", line 251, in dumps
    sort_keys=sort_keys, **kw).encode(obj)
  File "/usr/lib/python2.7/json/encoder.py", line 209, in encode
    chunks = list(chunks)
  File "/usr/lib/python2.7/json/encoder.py", line 442, in _iterencode
    o = _default(o)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1436, in default
    return flask.json.JSONEncoder.default(self, obj)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/json.py", line 83, in default
    return _json.JSONEncoder.default(self, o)
  File "/usr/lib/python2.7/json/encoder.py", line 184, in default
    raise TypeError(repr(o) + " is not JSON serializable")
TypeError: Undefined is not JSON serializable
2018-08-03 16:01:38,774 - octoprint.server.preemptive_cache - INFO - ... done in 7.14s
2018-08-03 16:01:40,907 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages/ (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-08-03 16:01:40,908 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-08-03 16:01:41,010 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
```

#### Link to contents of Javascript console in the browser

nothing 

I have read the FAQ.

foosel, 2018-08-03:
That looks like there are still left overs somewhere in the system from an earlier OctoPrint 1.4.0.dev install (see the `PERMISSIONS` bit).

Please try

```
cd ~/OctoPrint
source ./venv/bin/activate
python setup.py clean
pip install .
```

gege2b, 2018-08-03:
you're right, I was on the commit channel but for some reason that did't work, then I did some channel  switches and I clearly broke something

At first there was only the sarge 0.1.5 problem but once this problem solved, something was still broken

So,thank you now it works

I leave, I have to slap my head with something hard while repeating "you-will-not-mess-again-with-the-packages"

closed 2018-08-03 by gege2b

foosel, 2018-08-03:
Don't beat yourself up over it ;)