OctoPrint/OctoPrint #2780: Autoreport settings failure

celogeek, 2018-08-04:
#### What were you doing?
Changing the temperature autoreport interval in the settings from 2 to 5 or 10

1. ... Settings
2. ... Serial Connection
3. ... Intervals & Timeouts
4. ... Temperature interval (autoreport)
5. ... Set to 5
6. ... Save

#### What did you expect to happen?
Save and close the settings without errors. auto report interval change to 5s (M155 S5)

#### What happened instead?
Settings stay open.
This error occur in the log:
```
2018-08-04 08:31:37,135 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 294, in setSettings
    response = _saveSettings(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 418, in _saveSettings
    if "timeoutTemperatureAutoreport" in data["serial"]: s.setFloat(["serial", "timeout", "temperatureAutoreport"], data["serial"]["timeoutTemperatureAutoreport"], minimum=0.0)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 1688, in setFloat
    self.set(path, floatValue, **kwargs)
TypeError: set() got an unexpected keyword argument 'minimum'
2018-08-04 08:31:37,141 - tornado.access - ERROR - 500 POST /api/settings (::ffff:192.168.200.8) 24.39ms
2018-08-04 08:34:48,675 - octoprint.server.heartbeat - INFO - Server heartbeat <3
```

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.9

#### Operating System running OctoPrint
Linux Octopi 0.15.0

#### Printer model & used firmware incl. version
Tevo Tornado, Marlin bugfixes-2.0.x
It fail with tornado connected or not.

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

kazibole, 2018-08-05:
I am also experiencing the same problem while trying to change settings on that page, like:
- Temperature interval (polling)
- SD status interval (polling)
- Communication timeout

Running OctoPrint 1.3.9 on OctoPi 0.15.1. Also tried in safe mode and the problem remains.

The following error occurs in the Octoprint log:

```
2018-08-04 20:53:50,825 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 294, in setSettings
    response = _saveSettings(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 419, in _saveSettings
    if "timeoutSdStatus" in data["serial"]: s.setFloat(["serial", "timeout", "sdStatus"], data["serial"]["timeoutSdStatus"], minimum=1.0)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 1688, in setFloat
    self.set(path, floatValue, **kwargs)
TypeError: set() got an unexpected keyword argument 'minimum'
2018-08-04 20:53:50,831 - tornado.access - ERROR - 500 POST /api/settings (::ffff:172.16.12.20) 29.13ms
```

Carpedraconis, 2018-08-12:
I'm having the same issue, except it happens upon booting the pi.  This system worked great until a few weeks ago until I shut it down and unplugged to move to a new location.  Today, I get it set up in the new location and power on, and nothing will work.  Once the pi boots, it gives only 500 errors after a 60 second delay. Investigation show this happen in the log during startup.

```
2018-08-12 18:25:58,995 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://octopi.jam/'}
2018-08-12 18:26:04,103 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2018-08-12 18:26:05,101 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2018-08-12 18:26:56,758 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 399, in index
    response = default_view()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 385, in default_view
    return preemptively_cached()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 887, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 941, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 674, in decorated_function
    rv = f_with_duration(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 643, in f_with_duration
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 896, in decorated_function
    rv = f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 912, in decorated_function
    rv = f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 372, in make_default_ui
    r = make_response(render_template("index.jinja2", **render_kwargs))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/templating.py", line 128, in render_template
    context, ctx.app)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/templating.py", line 110, in _render
    rv = template.render(context)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/jinja2/environment.py", line 989, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/jinja2/environment.py", line 754, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/templates/index.jinja2", line 149, in top-level template code
    {% include 'javascripts.jinja2' %}
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/templates/javascripts.jinja2", line 4, in top-level template code
    {% assets "js_core" %}<script type="text/javascript" src="{{ ASSET_URL }}"></script>{% endassets %}
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/ext/jinja2.py", line 186, in _render_assets
    urls = bundle.urls()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 806, in urls
    urls.extend(bundle._urls(new_ctx, extra_filters, *args, **kwargs))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 765, in _urls
    *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 619, in _build
    force, disable_cache=disable_cache, extra_filters=extra_filters)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 543, in _merge_and_apply
    kwargs=item_data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/merge.py", line 276, in apply
    return self._wrap_cache(key, func)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 219, in fixed_wrap_cache
    content = self.cache.get(key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 182, in fixed_get
    hash = make_md5(self.V, key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/cache.py", line 84, in make_md5
    for d in walk(data):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/cache.py", line 61, in walk
    for d in walk(item): yield d
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/merge.py", line 79, in data
    return f.read()
  File "/home/pi/oprint/lib/python2.7/codecs.py", line 314, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xa0 in position 4167: invalid start byte
2018-08-12 18:2
6:56,834 - octoprint.server.preemptive_cache - INFO - ... done in 57.84s
```
Nothing changed from what was working a few weeks ago.  When booting, I did see github called, so it is possible it downloaded a new patch?

foosel, 2018-08-13:
@Carpedraconis that is *not* the same issue. Your's looks like your file system got corrupted. Never just pull power on the Pi, always shut it down properly first.

Carpedraconis, 2018-08-14:
I'm a Linux admin, I know better.  I'm getting the same python code error
on the same line as the other entries.

foosel, 2018-08-15:
@Carpedraconis this issue is about what looks to be a keyword issue inside OctoPrint's code, triggered when attempting to save one of the timeout options in the settings dialog. Your issue is about some mangled data inside the webasset cache triggered when attempting to access the UI. That's not even close "the same python code error".

Carpedraconis, 2018-08-15:
Same line of code, same file, same stack trace puts both issues in the same
part of the code.  I completely removed the web cache and rebooted the pi
(so it would rebuild) and the problem persists.  So no, it's not a "user
screwed up the cache" issue.

foosel, 2018-08-16:
I didn't say anything about "the user screwed up", is said it looks like some corrupt cached file.

The stack trace from the OP:

```
2018-08-04 08:31:37,135 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 294, in setSettings
    response = _saveSettings(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 418, in _saveSettings
    if "timeoutTemperatureAutoreport" in data["serial"]: s.setFloat(["serial", "timeout", "temperatureAutoreport"], data["serial"]["timeoutTemperatureAutoreport"], minimum=0.0)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 1688, in setFloat
    self.set(path, floatValue, **kwargs)
TypeError: set() got an unexpected keyword argument 'minimum'
```

and the stack trace from @kazibole 

```
2018-08-04 20:53:50,825 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 294, in setSettings
    response = _saveSettings(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 419, in _saveSettings
    if "timeoutSdStatus" in data["serial"]: s.setFloat(["serial", "timeout", "sdStatus"], data["serial"]["timeoutSdStatus"], minimum=1.0)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 1688, in setFloat
    self.set(path, floatValue, **kwargs)
TypeError: set() got an unexpected keyword argument 'minimum'
```

Your stack trace:

```
2018-08-12 18:26:56,758 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
[... 45 lines cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/bundle.py", line 543, in _merge_and_apply
    kwargs=item_data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/merge.py", line 276, in apply
    return self._wrap_cache(key, func)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 219, in fixed_wrap_cache
    content = self.cache.get(key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 182, in fixed_get
    hash = make_md5(self.V, key)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/cache.py", line 84, in make_md5
    for d in walk(data):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/cache.py", line 61, in walk
    for d in walk(item): yield d
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets/cache.py", line 61, in walk
    for d in walk(item): yield d
[... cut ...]
UnicodeDecodeError: 'utf8' codec can't decode byte 0xa0 in position 4167: invalid start byte
2018-08-12 18:2
```

How is this the same file, the same line of code, the same endpoint, the same traceback, the same anything? It's not the same source file, it's not the same call that led to it, it's *completely unrelated to this ticket*. I'm not saying this isn't an issue or that you are who caused it, but it has nothing to do with this ticket here. Please open another ticket, include all required information (full logs, reproduction steps, etc) so this can be looked into properly.

ManuelMcLure, 2018-08-16:
I think the confusion here is that python dumps its tracebacks in the opposite order than is common for other languages (i.e. C/C++). As a primarily C/C++ developer, this was a point of confusion for me when I first started looking at python tracebacks. If you're a C/C++ developer, your eye automatically goes to the first function listed when comparing tracebacks instead of the last function as is produced by python.

foosel, 2018-08-17:
@ManuelMcLure That's my impression as well. I have a couple of languages under my belt and am aware how much such differences can throw one off. But if I'm not familiar with a language, and someone who is tells me that I'm misinterpreting a traceback, then I listen the first time instead of doubling down and becoming passive aggressive 🤷

foosel, 2018-08-20:
Fixed by the above commit, already available on `maintenance`, soon `devel`, will be released as part of 1.3.10.

celogeek, 2018-08-22:
thanks ! 👍 
when do the rc/maintenance will be updated ?

foosel, 2018-08-22:
When it's time to push out a first RC for 1.3.10, for which I don't have a date yet ;)

Boby71, 2018-10-04:
Hmmm...running 1.3.9 (stable) and get the same error when trying to change the timeouts:
"TypeError: set() got an unexpected keyword argument 'minimum'"
Any idea how to get rid of it?

kazibole, 2018-10-05:
> Hmmm...running 1.3.9 (stable) and get the same error when trying to change the timeouts:
> "TypeError: set() got an unexpected keyword argument 'minimum'"
> Any idea how to get rid of it?

It will be fixed in 1.3.10.

Boby71, 2018-10-05:
Thanks, @kazibole - I was confused by the „solved“ tag of this issue.

closed 2018-12-10 by foosel

hYdos, 2019-04-01:
i got this problem but i cant read the sd card to get the logs because i broke my sd card reader

commit 26856d05 (2018-08-20): Fix wrong min/max parameter being used for timeouts

Closes #2780