OctoPrint/OctoPrint #2781: [Request]

shredda, 2018-08-04:
Currently I am using OctoPrint 1.3.8 running on OctoPi 0.15.1 on a Raspberry Pi 3.

I a very pleased with the features and operation of octoprint and find it a fantastic project.

It would be great to have a possibility to abort newly started prints. But for prints that are already running several minutes or even hours, the confirmation box should appear.

Something like
[] Confirm before cancelling a print
[] Confirm before cancelling a print running for more than [n] layers / [n] minutest

For me both layer and time definition of the "need cancellation confirmation"-point would be fine.

Then I would set the limit to something like 3 layers and can then abort prints going wrong in their early stages easily without the need to fear an accidental abort of a print running for hours.

foosel, 2021-04-07:
@LazeMSS not sure what plugin you tried to link there 🤔

LazeMSS, 2021-04-07:
https://plugins.octoprint.org/plugins/inlineconfirm/

Oops