OctoPrint/OctoPrint #2782: shredded moving while printing circle

posekac, 2018-08-04:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

posekac, 2018-08-04:
Always when printing circles or round lines it makes shreded moves and the extruded filament acumulates on one spot so the printed thing doesnt look good. when printing from SD card it is OK. So it must be problem with octoprint.

Version of OctoPrint 1.3.9
Original prusa i3 MK3 fw 3.3.1

https://www.youtube.com/watch?v=P-pDhwOrN5Q

tdub415, 2018-08-04:
This is, from what I have experienced and can tell, due to the limit of how much data can be transmitted to the printer via USB, as opposed to SD.

posekac, 2018-08-04:
The problems startred after downloading latest update few days ago. First I thought it was problem with GCode than I tried to print some other simply prints but all the same. Only printing from SD card was ok.

foosel, 2018-08-06:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

jrdiaz, 2018-08-07:
I started suffering from this issue since last update.
The print staggers, shutters and has hiccups randomly pausing when printing small segments creating blobs and infimal quality. Besides that prints lasts now nearly the double of time but the read issue is the huge quality drop to unusable levels.
Printing same gcode from SD card turned out perfect prints.
The smaller the part is, the uglier it looks.

Straight lines are ok, but arcs are impossible. As an example, printing the brim for a 30cm diameter cylinder suffers from pauses since the first moment pausing in each brim circle arround 20 times.

Hope that this helps diagnosing the problem.

Original Prusa MK2S
Octoprint 1.3.9 version

foosel, 2018-08-07:
I can't look into this with the *requested information* - **logs**, information whether the issue persists in **safe mode**, test files, ...

closed 2018-08-15 by foosel