OctoPrint/OctoPrint #2784: Error installing netifaces during 1.3.8 to 1.3.9 upgrade

aaronbieber, 2018-08-05:
#### What were you doing?

1. Upgrading from 1.3.8 to 1.3.9 using the update function
2. ...
3. ...

#### What did you expect to happen?

OctoPrint 1.3.9 would be installed.

#### What happened instead?

The upgrade failed while attempting to install the `netifaces` Python library.

#### Did the same happen when running OctoPrint in safe mode?

The run mode of OctoPrint has no effect on the upgrade process.

#### Version of OctoPrint

1.3.8

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

Prusa i3 MK2S

Provided firmware, no modifications.

#### Browser and version of browser, operating system running browser

Firefox 62.0b10, macOS 10.13.6

#### Link to octoprint.log

https://gist.github.com/aaronbieber/9e7cc1863b85953d13ba879f21c00c9f

#### Link to contents of terminal tab or serial.log

`plugin_softwareupdate_console.log`:

https://gist.github.com/aaronbieber/0c0f867f2a37539171f35f9491057655

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-10-29:
Looks like this one slipped through, sorry.

I just tried to this against OctoPi 0.13 with an OctoPrint updated to 1.3.8 and I couldn't reproduce it:

```
2018-10-29 12:39:13,548 > Running setup.py install for netifaces: started
2018-10-29 12:39:32,502 > Running setup.py install for netifaces: finished with status 'done'
[...]
2018-10-29 12:39:54,410 > Successfully installed Flask-Login Jinja2 OctoPrint backports-abc netaddr netifaces pkginfo psutil pyserial rsa singledispatch tornado
```

Update went through flawlessly. So it doesn't appear to be a general issue. 

It might be something else in your environment causing this, C compiler version or something, also possibly a file system corruption, but I fear that reproducing this might proof tricky to impossible. The error also occurs while installing a third party dependency, not within OctoPrint itself, so other than possibly changing the pulled in version there's not a thing I could to to solve this.

Can you compile C programs in general? Have you ever just pulled the plug on this Pi in the past without shutting it down first? Does this happen reliably on attempts to install 1.3.9?

aaronbieber, 2018-10-29:
Thanks for taking a look at this, @foosel; I can take a closer look later when I am back at home, but this was an OctoPi image originally and I haven't mucked about with any of the build tools or anything. The crash does happen reliably with every attempt to install 1.3.9.

I have installed other plug-ins and upgraded OctoPrint at least once before, which would potentially make my system differ from a clean install of a version 1.3.8 OctoPi.

If I wanted to start over from a fresh disk image, is there a straightforward way to back up my many configurations? I should like to be able to restore the configurations of my plugins as well, if that is possible.

I happen to have a couple of other RaspberryPis laying around (who doesn't, right?) so I could try a fresh install there and try to migrate my data to get things back into a working state. If this problem isn't happening to anyone else it is probably some quirk of the upgrade paths I have taken in the past and I'll bite the bullet and start fresh.

foosel, 2018-10-29:
> If I wanted to start over from a fresh disk image, is there a straightforward way to back up my many configurations? I should like to be able to restore the configurations of my plugins as well, if that is possible.

Settings and files are not an issue, installed plugins (not their settings, but their actual code) is a bit manual work right now. See [this FAQ entry](https://discourse.octoprint.org/t/how-do-i-backup-my-octoprint-settings-on-octopi/1489). As it states there, with 1.3.10 that will all become a bit more straight forward thanks to the bundled backup plugin.

If you have multiple Pis available simply setting up a fresh install on another one and moving over `~/.octoprint` might indeed be the easiest way. 

I so far am not aware of any other instances of this problem.

foosel, 2019-04-05:
Is this still an issue?

aaronbieber, 2019-04-05:
I haven't re-attempted, I'm fine with closing it. Thanks for your help!

closed 2019-04-05 by aaronbieber