OctoPrint/OctoPrint #2785: Dependencies broken, can't install

radry, 2018-08-05:
#### What were you doing?
Installing octoprint

1.Installing Python 2.7 including pip and virtualenv.
2.Obtaining the source through cloning the source repository: git clone https://github.com/foosel/OctoPrint.git  
3. Creating a (user owned) virtual environment in the source folder: virtualenv venv
4. Installing OctoPrint into that virtual environment: ./venv/bin/python setup.py install

Following the guide [here ](https://discourse.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337)

#### What did you expect to happen?

#### What happened instead?

When running `./venv/bin/python setup.py install`  following error appears:  
`Processing dependencies for OctoPrint==1.3.9
Searching for frozendict<1.3,>=1.2
Reading https://pypi.python.org/simple/frozendict/
Download error on https://pypi.python.org/simple/frozendict/: unknown error (_ssl.c:2947) -- Some packages may not be found!
Couldn't find index page for 'frozendict' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: unknown error (_ssl.c:2947) -- Some packages may not be found!
No local packages or working download links found for frozendict<1.3,>=1.2
error: Could not find suitable distribution for Requirement.parse('frozendict<1.3,>=1.2')`  

When following the official guide [here ](https://discourse.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337) following error appears:  
`(venv) octoprint@orangepizero:~/OctoPrint$ pip install https://get.octoprint.org/latest
Collecting https://get.octoprint.org/latest
Exception:
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/home/octoprint/OctoPrint/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/octoprint/OctoPrint/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/octoprint/OctoPrint/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/octoprint/OctoPrint/venv/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/octoprint/OctoPrint/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
SSLError: unknown error (_ssl.c:2947)`

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.3.9

#### Operating System running OctoPrint
Armbian 5.38 (next)

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-08-06:
I cannot reproduce this here on a fresh install attempt. Based on the output:

    raise SSLError(e, request=request) SSLError: unknown error (_ssl.c:2947)

it looks like something is/was going wrong with the SSL connection to the python package index, which is also something I can't fix. Check that the system has the correct system date and time (otherwise SSL validation will go wrong) and that regular connections to https://pypi.python.org work fine. Also make sure that you are running a current version of `pip` just in case.

closed 2018-08-06 by foosel

radry, 2018-08-06:
>Check that the system has the correct system date and time  

System time and date is correct. I don't know what to do. I have a fresh install, new pip, patest packages and the install script fails.  

The error also says  
`No local packages or working download links found for frozendict<1.3,>=1.2 error: Could not find suitable distribution for Requirement.parse('frozendict<1.3,>=1.2')`  

So obviosuly the installation script is not satisfied with the packages it finds. Is there a way to manually install this dependency?

foosel, 2018-08-07:
It can't find the package because it runs into an error even talking to the package repository. You could try manually downloading all the dependencies, transferring them to your device and then installing them manually. But that's only a solution for the initial install and you'd probably run into issues on every single update thereafter.

Something in your system is preventing Python from making a connection to the package index. The question is what - "unknown error" sadly doesn't help a whole ton and a quick googling around also doesn't really suggest anything but system date and time or alternatively outdated system certificate chains (try `sudo update-ca-certificates` maybe?)

Robirt55, 2018-11-30:
I was experiencing this error. I found this post pretty quick googling. Then attempted to run the  `sudo update-ca-certificates` command. This corrected the issues. 

Thanks!

jebbam, 2018-12-12:
I saw a similar issue using a fresh install of Debian Stretch on a BBB, using BBB's image.

The error:

```
Installing octoprint script to /usr/local/bin
Installed /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.10-py2.7.egg
Processing dependencies for OctoPrint==1.3.10
Searching for frozendict>=1.2,<1.3
Reading http://pypi.python.org/simple/frozendict/
Couldn't retrieve index page for 'frozendict'
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for frozendict>=1.2,<1.3
error: Could not find suitable distribution for Requirement.parse('frozendict>=1.2,<1.3')
```

Note that the download URL that was tried was http not https. This caused the download to fail, because pypi doesn't permit index viewing using http. The simple solution was to install python-openssl, which added SSL support so it uses https instead.