OctoPrint/OctoPrint #2787: Communication with printer freezes

JoergGu, 2018-08-08:
#### What were you doing?
I upgraded the Firmware on my Creality CR-10S (Board Rev. 2.1) to Marlin 1.1.9 and tried to establish a connection.

#### What did you expect to happen?
 A functional connection.

#### What happened instead?
Now the printer freezes after connecting to it.
The display of the CR-10s gets blank.
After a while the status of OctoPrint changes to:
Offline (Error: Too many consecutive timeouts, printer still connected and alive?)

Sometimes that happens already when "Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125" is displayed in the terminal, as previosly mentioned in issue #1978.
(I tried the solutions that were provided in #1978, but it didn't solve the problem)
Nevertheless, today I cannot reproduce that behaviour. Maybe later.

#### Did the same happen when running OctoPrint in safe mode?
the same problem

#### Version of OctoPrint
OctoPrint 1.3.9

#### Operating System running OctoPrint
OctoPi 0.13.0

#### Printer model & used firmware incl. version
Creality CR-10S (Board Rev. 2.1), Marlin 1.1.9

#### Browser and version of browser, operating system running browser
FF 61.0.1, Win 10

#### Link to octoprint.log
https://gist.github.com/JoergGu/06f1e914cfb23f8adc07146af9fb7d5d#file-octoprint-log

#### Link to contents of terminal tab or serial.log
https://gist.github.com/JoergGu/06f1e914cfb23f8adc07146af9fb7d5d#file-terminal-txt

#### Link to contents of Javascript console in the browser
https://gist.github.com/JoergGu/06f1e914cfb23f8adc07146af9fb7d5d#file-javascriptconsole-txt

#### Screenshot(s)/video(s) showing the problem:

[image]

foosel, 2018-08-08:
Looks like your firmware gets stuck while reading from *the printer's* SD card. Try removing it, then reconnecting. If it connects through then, the card's broken in some way. Otherwise please provide a new `serial.log` with the card removed.

foosel, 2018-08-08:
As a side note, if stuff stops working after you upgrade the printer's firmware, report the issue in the *firmware's issue tracker*.

JoergGu, 2018-08-08:
Ok, that seems to solve the problem - rebootet three times without a communication failure.
Nevertheless it's quite interesting because running Marlin 1.1.8 and the original Creality-Firmware the problem isn't present.

By the way, before removing the SD card, I was able to reproduce the behavior mentioned in #1978, that causes the same communication freeze.
I have attached corresponding files below.
Maybe it's the same reason (Problem with SD card) even though it stops earlier - so those files maybe obsolete and only for documentation purposes.

Here is a new terminal.log:
https://gist.github.com/JoergGu/06f1e914cfb23f8adc07146af9fb7d5d#file-terminal2-txt

And an updated octoprint.log:
https://gist.github.com/JoergGu/06f1e914cfb23f8adc07146af9fb7d5d#file-update_octoprint-log

THX for the quick response and solution. :-)

foosel, 2018-08-08:
> Nevertheless it's quite interesting because running Marlin 1.1.8 and the original Creality-Firmware the problem isn't present.

Whatever changed between 1.1.8 and 1.1.9 apparently made the printer crash on reading that specific card then. Is this an official Marlin build or something from Creality? If the former, I suggest to also report this in the Marlin bug tracker, just in case this is indeed a regression on their end.

I'm closing this since it was indeed caused by the presence of the SD card which for some reason confused the firmware.

closed 2018-08-08 by foosel