OctoPrint/OctoPrint #2794: Print Stuttering when loading Web UI with Pi 3B+

ErinTheSmall, 2018-08-18:
#### What were you doing?
Loading the Web UI

1. ... Have A USB webcam connected
2. ... Have a print with complex geometry running
3. ... Load web interface

#### What did you expect to happen?
 The web interface to load and the print to continue smoothly
#### What happened instead?
 The web interface loaded fine but the print stuttered and then continued
(however CPU usage didn't seem to spike much over 1.0 total when checking with the "W" command)
#### Did the same happen when running OctoPrint in safe mode?
 Yes

#### Version of OctoPrint
OctoPrint 1.3.9 

#### Operating System running OctoPrint
OctoPi 0.15.1

#### Printer model & used firmware incl. version
Custom Marlin based printer (Marlin 1.1.9)
(Raspberry pi 3b+ using inbuilt wifi)

#### Browser and version of browser, operating system running browser
Google Chrome + Windows 10 professional 

#### Link to octoprint.log
[Octoprint.log](https://gist.github.com/ErinTheSmall/62e44524a46661aeb2447a3ec452745d)

#### Link to contents of terminal tab or serial.log
 [Serial.log](https://gist.github.com/ErinTheSmall/c040cfd3675e327f4f9a4737ffae46b4)

#### Link to contents of Javascript console in the browser
[Javascript Console text](https://gist.github.com/ErinTheSmall/59b5a7475878427235f1a47ea8355fd3)
[Javascript Console image](https://drive.google.com/file/d/1Ma72wYMw9yTIg37RQ75I9K7tmi5sMPPt/view?usp=sharing)

#### Screenshot(s)/video(s) showing the problem:
[Video of the issue from camera and screen capture of Web UI](https://youtu.be/Y0vZjAknwxk)

I have read the FAQ.

ErinTheSmall, 2018-08-18:
Just tested without a webcam and with the official Pi power supply: I still get exactly the same outcome.

blaskkaffe, 2018-08-19:
I have had the same issue, the print momentarily stops when the webcam is loaded, ether on web interface, in cura or in the astroprint app when using the astroprint plugin. 

On web interface it only happens when i start a new session, not if i already have the tab open and running.

akatrevorjay, 2018-08-19:
Honestly the root of this problem is that octoprint runs only a single process to do, well, almost everything, ie web interface, printer communication, backend tasks, etc.

fiveangle, 2018-08-19:
Dupe of #1241

ErinTheSmall, 2018-08-19:
I don't think so..? @fiveangle  I have never done a timelapse on this pi.  there is no data to fail at deleting?

ErinTheSmall, 2018-08-19:
I do't think so? @fiveangle 
the page load is not long at all. 
[Screenshot](https://imgur.com/a/Mrm3pjL)

foosel, 2018-08-20:
Yep, duplicate.

closed 2018-08-20 by foosel