OctoPrint/OctoPrint #2795: File.open fails when opening file from sd card

dutchrazor, 2018-08-19:
#### What were you doing?

I have uploaded a file to the SD card of the Arduino, and wanted to load the print via the user interface. This workflow has worked sometime but stopped working a while back, I believe after I updated Octoprint to 1.3.6  (from 1.3.4?)

#### What did you expect to happen?

Normally the file is loaded into Repetier and the printer starts printing, with Octopi displaying the amount of bytes left.

#### What happened instead?

Send: N8 M23 /kap deksel.gcode*27
Recv: ok 8
Recv: file.open failed

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

Octopi version 0.14.0

#### Printer model & used firmware incl. version

DIY, using Repetier 1.0 (feb 2017)

#### Browser and version of browser, operating system running browser

Mac using Safari, also tested on Windows with Chrome

#### Link to octoprint.log

https://gist.github.com/dutchrazor/72026d37950ede25792fbbea66efd3cd

#### Link to contents of terminal tab or serial.log

Recv: Printed filament:5807.62m Printing time:55 days 10 hours 13 min
Recv: PrinterMode:FFF
Recv: ok 7
Recv: wait
Recv: T:29.09 /0 B:46.28 /0 B@:0 @:0
Recv: wait
Recv: T:29.09 /0 B:45.90 /0 B@:0 @:0
Send: N8 M23 /kap deksel.gcode*27
Recv: ok 8
Recv: file.open failed
Recv: T:29.09 /0 B:45.60 /0 B@:0 @:0
Recv: wait
Recv: T:29.09 /0 B:45.45 /0 B@:0 @:0
Recv: wait
Recv: T:29.09 /0 B:45.75 /0 B@:0 @:0

ErinTheSmall, 2018-08-19:
Try updating your octoprint version to the latest?

foosel, 2018-08-20:
```
Recv: file.open failed
```

This is your *printer* stating that it can't open the file to print it. That's nothing OctoPrint can do anything about. Might be that your printer's SD card is at its end of life. Try swapping the card. In any case, not a bug in OctoPrint.

closed 2018-08-20 by foosel

RobinTail, 2018-10-31:
I have the same issue. Octoprint is running on Raspberry Pi 3B+, connected to Anet A6.
There is a brand new readable SD card installed in 3d printer with a plenty of gcode files written using my MacBook card reader. Printer itself can read and print these files from installed card, but I can not choose them in Octoprint. In log file I can see the same message.
I also confused about filenames. After little investigation I found that only files that were uploaded to printer's SD card using Octoprint can be opened by Octoprint. 

[image]