OctoPrint/OctoPrint #2798: [FEATURE REQUEST] override sd upload serial by plugin

devildant, 2018-08-24:
Hello, 
I develop an octoprint plugin to use the flashcard.
currently I can have something "functional".
I can download a 150 MB file in 4 ~ 5 min.
However, I do not find a solution for my plugin, to replace the upload system SD octoprint (just upload).
I search in the code octoprint and I did not find elements allowing to use a plugin instead of the serial upload.
I also work on an electronic solution, to make the upload on SD much faster.
would it be possible to have a way to overload the serial upload with that of a plugin?

best regards

devildant, 2018-09-13:
nobody ?

foosel, 2018-09-14:
Got my hands terribly full these days, sorry. 

In short though: Currently not possible to replace the sdcard storage implementation through a plugin (at least not as far as I am aware of).

devildant, 2018-10-13:
Hello @foosel ,
I am only transmitting my progress.
currently my module, copy to the SD card of the printer with speeds between 25 and 50Mo/s (usually 25: p)
It takes about 10 seconds to copy a 150 Mo file to the SD card.
I'm currently working on modifying octoprint to incorporate the necessary changes.
However, my understanding of the octoprint remains basic, could you look and help me when I will have something functional?
for a future pull request: p

foosel, 2018-10-15:
> However, my understanding of the octoprint remains basic, could you look and help me when I will have something functional?

Sorry, but I have my hands already horribly full, I can't commit to this kind of coaching at the moment :/

ringerc, 2019-02-28:
I find the SD card speed surprising TBH.

Sure, usb-serial is a hideous, slow and primitive channel, and overlaying G-code over it makes it worse still. I assume there's something forcing a relatively slow RTT for each small chunk of data, and firmwares don't support anything like a g-code command to switch into the "good" old ZModem protocol to do a more efficient load or unload.

It'd be good to improve the UI a little to warn that SD upload is very slow when the user attempts it. I tried what I thought was a small SD upload of a 15MB file and it was going to take 20 minutes, so I just cancelled it (about which I have a bug report coming soon) and copied straight to the SD card ... only to hit bug #1600. Oh well.

SD support aside it's fantastic :)

devildant, 2019-02-28:
hello, i am currently working on a solution it should arrive soon

cp2004, 2022-08-02:
Can be done with the hook [`octoprint.printer.sdcardupload`](https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-printer-sdcardupload) since 1.3.11.

closed 2022-08-02 by cp2004

devildant, 2022-08-02:
yes I know it's me who implemented it, and I use it every day ^^