OctoPrint/OctoPrint #2803: Octoprint crash while printing

marwatk, 2018-09-01:
#### What were you doing?
About 2 hours into a print. Octoprint window open in browser, cura open monitoring Octoprint

Unknown, waiting on next print. My guess is it's a transient issue, though.

#### What did you expect to happen?

Print completes

#### What happened instead?

Octoprint crash with stack trace:
```2018-09-01 17:40:57,729 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.128
2018-09-01 17:40:58,865 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.128
2018-09-01 17:44:08,628 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-09-01 17:45:16,566 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 1639, in _monitor
    self._handle_ok()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2087, in _handle_ok
    self._continue_sending()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2197, in _continue_sending
    if self._send_from_command_queue():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2375, in _send_from_command_queue
    entry = self._command_queue.get(block=False)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3865, in get
    return TypedQueue.get(self, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1401, in get
    item, _ = PrependableQueue.get(self, *args, **kwargs)
  File "/usr/lib/python2.7/Queue.py", line 182, in get
    self.not_empty.release()
TypeError: 'thread.lock' object is not callable
2018-09-01 17:45:16,613 - octoprint.plugins.pushover - INFO - Event triggered: Error 
2018-09-01 17:45:16,660 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: See octoprint.log for details)"
2018-09-01 17:45:16,678 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 13.18ms
2018-09-01 17:45:16,724 - octoprint.filemanager.analysis - INFO - Starting analysis of local:CCR10_Drift_Mask_2_3_part_1.gcode
2018-09-01 17:45:16,742 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 11.17ms
2018-09-01 17:45:16,879 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 11.35ms
2018-09-01 17:45:17,537 - octoprint.plugins.pushover - INFO - Event triggered: PrintFailed 
2018-09-01 17:45:17,982 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 10.57ms
2018-09-01 17:45:17,998 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 11.39ms
2018-09-01 17:45:18,016 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 12.16ms
2018-09-01 17:45:18,031 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.128) 10.31ms
```

#### Did the same happen when running OctoPrint in safe mode?

Waiting on next print to see if I'm able to duplicate running normally.

#### Version of OctoPrint

OctoPrint 1.3.9

#### Operating System running OctoPrint

Octopi 0.15.1
Linux octopi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

#### Printer model & used firmware incl. version

Ender 3, Marlin Stock 1.1.8 with BLTouch enabled

#### Browser and version of browser, operating system running browser

Chrome Version 68.0.3440.106 (Official Build) (64-bit) Windows 10

#### Link to octoprint.log

https://gist.github.com/marwatk/e30f132bcbc11738b4313cd9c1201245

#### Link to contents of terminal tab or serial.log

I tried to refresh the page before capturing the terminal tab, if it happens again I will update this issue.

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.
Yes

foosel, 2018-09-21:
Just to verify, has this shown up another time? I'm a bit confused since on first look it looks like it's triggered inside Python's own queue implementation, so that's kinda unexpected...

marwatk, 2018-09-21:
Nope, haven't been able to duplicate it...

foosel, 2018-10-18:
Let's leave it open but mark as unreproduced then for now... Maybe it happens for someone else and we get some more data that helps in narrowing things down.

amoose136, 2018-11-15:
Was the pi otherwise responsive after the crash? I just had a print get 74% complete before failing and when I came to the Pi it was hard crashed to the point that tapping the touchscreen would not wake it up and I couldn’t communicate with it from a desktop. I’m trying to figure out if this is the same issue or something different.

foosel, 2018-11-15:
@amoose136 that doesn't sound like this issue, or any issue in OctoPrint actually, but rather something that's wrong on an OS or hardware level.

amoose136, 2018-11-15:
I just checked octoprint.log. It's a plugin issue I think. Even though I didn't touch the server during the printing process it seems the Exclude Region plugin had some issues and so did Cancel Object and this was followed by a hard crash. The server restarted but then appears to have frozen some time later creating a 6 hour gap in the logs.

b14ckyy, 2018-12-26:
Hi there. Just now it happened to me too (again). I had an issue open a few months ago (#2329) but since then it never happened again. At this time I was able to see it happen. The printer just stops and then Marlin (1.1.8) just resets. Web Interface was open, no other application was active (Cura was closed and Printoid not started).

Octoprint Version 1.3.10, Image 0.15.1.

I´m still able to access the core system via SSH and WinSCP, so I was able to download the actual Octoprint log (that only stops at 22:19) and also the system logs that are updated short before the crash. 

Octoprint is fresh installed a few days before with all plugins installed I normally use. This was not happening a few months now. It is very rare so trying to reproduce in Safe mode is hard. And Analyze if a plugin causes this is nearly impossible, at least to identify it. It also happened without the web interface open in the past. Had this issue since nearly a year now (since 1.3.6 the first time).

Logs attached. 
[Logs.zip](https://github.com/foosel/OctoPrint/files/2710903/Logs.zip)

foosel, 2019-11-05:
Still haven't seen this happen myself and sadly also not seen anything in logs.

Should anyone run into this with current versions, please
 
  * Share `octoprint.log`
  * Share `serial.log` (needs to be enabled first) or terminal tab contents (just in case it's related to something the printer sent)
  * Share list of installed plugins & versions

b14ckyy, 2019-11-05:
did not see it too since spring.... maybe it was a faulty plug in... idk. I had serial log turned on for 6 months but as it impacts my print speed i turned it on and gave up to reproduce.

LazeMSS, 2021-03-31:
Closing... No new problems since 2018

closed 2021-03-31 by LazeMSS