OctoPrint/OctoPrint #2810: Can't login on octoprint locally but working fine with printoid

Vanpourix, 2018-09-05:
Hi, I have an issue when attempt to log in my octoprint server locally.

#### What were you doing?

- 1. Installed octoprint completely including "local login" (During first start guide)
- 2. Installed some plugins (Telegram, printed history, print stats, cost estimation, ...)
- 3. Tested some prints locally (worked)
- 4. Port forwarded my octoprint (local: 80, public: 5000)
- 5. Set up the raspberry and printoid pro with Basic Authentication and tested distant print (Worked perfectly)
- 6. Connect on local (192.168.X.XX), it asked for the "public network authentication".
- 7. Try to login to my local profile (NOT WORKING)
- 8. Did tried to reinstall all the server, tried this too (both not worked): [](https://discourse.octoprint.org/t/i-forgot-my-octoprint-password-how-can-i-reset-it/215)
- Can ssh to 192.168.1.46 as "pi" (root admin). But when running commands, this message also appear "sudo: unable to resolve host octopi: No such file or directory"

#### What did you expect to happen?
To log in my octoprint profile in order to fully manage my server locally.

#### What happened instead?
Error message : " Login failed User unknown or wrong password " despite right credentials provided

#### Did the same happen when running OctoPrint in safe mode?
The same thing happen with safe mode.

#### Version of OctoPrint
Octoprint 1.3.8 running on OctoPi 0.15.1

#### Operating System running 
Octopi on raspberry pi 3B+

#### Printer model & used firmware incl. version
Anet A8 on latest marlin version

#### Browser and version of browser, operating system running browser
Firefox quantum 61.0.2.
(Tried with IE as well)

#### Link to contents of terminal tab or serial.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/2354386/octoprint.log)

#### Link to contents of terminal tab or serial.log
[serial.log](https://github.com/foosel/OctoPrint/files/2354393/serial.log)

#### Link to contents of Javascript console in the browser
I have 7 potential javascript in my firefox developer console that might be relevant, which one should I share ?

#### Screenshot(s)/video(s) showing the problem:
ø

I have read the FAQ.

ntoff, 2018-09-05:
>I have 7 potential javascript in my firefox developer console that might be relevant, which one should I share 

The whole thing, just take a screenshot (or copy / paste) of the whole console.

Vanpourix, 2018-09-06:
Thanks for your reply. I added a screenshot of the whole console.

foosel, 2018-09-07:
```
2018-09-05 19:33:06,738 - tornado.access - WARNING - 401 POST /api/login (127.0.0.1) 24.01ms
```

Looks like a wrong user name or password. Check caps lock, make sure you are typing everything exactly in the case it should be in, check that no trailing whitespace is attached anywhere, ...

If that doesn't help, please open a support request in the "Get Help" section on [the forums](https://discourse.octoprint.org), as this is likely an issue with your setup, not with OctoPrint itself - logins are working flawlessly over here.

> Port forwarded my octoprint (local: 80, public: 5000)

[Please stop doing that](https://octoprint.org/blog/2018/09/03/safe-remote-access/).

closed 2018-09-07 by foosel

Vanpourix, 2018-09-07:
As mentionned I am 100% sure of the credential provided (verified through the http headers & parameters). And the issue come back as well after password reset.
I think it might be a corruption between the multiple authentication and server setup so I am opening a support request. Would be the occasion to verify the happroxy setup as well.
Thanks for your answer.