OctoPrint/OctoPrint #2812: nginx octoprint

garffi, 2018-09-06:
Hello,

I have a small server octoprint on my machine that has a nginx accessible from the outside, with as configuration:

```
location /octo/ {
proxy_pass http://localhost:5000/; # make sure to add trailing slash here!
proxy_set_header Host $http_host;
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection "upgrade";
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Scheme $scheme;
proxy_set_header X-Script-Name /octoprint;
proxy_http_version 1.1;
client_max_body_size 0;
}
```
The problem is when I access the octo server on my external address. It does not load the js and css.
in the log nginx I'm like a line, the log nginx I was kind of line:
`*177 open() "/mnt/www/octoprint/static/webassets/packed_core.js" failed (2: No such file or directory)`

Does this mean that octoprint does not support being placed somewhere other than directly at the root of a domain, because paths to external resources (JS, CSS) are hardcoded?, this signifi qu'octoprint not stand to be placed other than directly to the root of a domain, because the paths to the external resources (JS, CSS) are hardcode?

What is the solution?

Erutan409, 2018-09-06:
@garffi Can you clarify the environment(s) you're running everything on/from?  Are you running a single web server (nginx) and have installed Octoprint in a sub-folder, etc.?

I'm running Octoprint on a raspberry pi from the image provided.  I'm also running nginx on another raspberry pi as reverse proxy server for multiple different systems I'm hosting from home.  I setup a reverse proxy to forward all traffic to my Octoprint instance for a specific sub-domain I have registered with my web host and I can attest to that working.  Even with SSL setup with Let's Encrypt.

Is that along the lines of what you're attempting to do?

garffi, 2018-09-06:
@ Erutan409 I use simple octoprint on an archlinux where the machine hosts nginx and the printer plugged into that same machine. 

> I'm also running nginx on another raspberry pi as reverse proxy server for multiple different systems I'm hosting from home.

Can you specify a little bit about your configuration and where do you think mine is planting? I actually use Let's Encrypt.

But with proxy_pass, nginx looks for js and css files to the path of my root www (/ mnt / www).

ntoff, 2018-09-06:
This is a bug tracker for issues, for help and support please use the discussion forum located at https://discourse.octoprint.org/

foosel, 2018-09-07:
As @ntoff said, this is a topic better tackled on the support forums (where the FAQ is also located btw which contains configuration examples for nginx as well).

closed 2018-09-07 by foosel