OctoPrint/OctoPrint #2816: Wrong printing percentage

diesphink, 2018-09-08:
#### What were you doing?

After a file size change of a loaded gcode file, printing percentage shows over 100% as it uses the old file size.

Steps to reproduce:
1. Load a model for printing;
2. Upload (via ssh in my case) a new model with a bigger file size, overriting loaded file;
3. Update files, it will show the new file with bigger file size on the list, but won't update the loaded information, as pictured:
[image]
4. During printing, octoprint shows bigger than 100% progress (and also maybe sends data faster than 
should? This part got a little bit weird, maybe it was just the gcode viewer that got lost)

#### What did you expect to happen?

After reloading the file list, if octoprint finds changed information on a loaded file, it should update the information.

#### Version of OctoPrint
OctoPrint 1.3.9

#### Operating System running OctoPrint
OctoPi 0.14.0 

#### Printer model & used firmware incl. version
Prusa i3 mk2.5

#### Browser and version of browser, operating system running browser
Firefox 61.0.1

#### Link to octoprint.log

https://gist.github.com/diesphink/7a7437a9e95dbe0cb1d34f108638ef5a

I have read the FAQ.

ntoff, 2018-09-09:
>Upload (via ssh in my case) a new model with a bigger file size, overriting loaded file;

Don't just upload an overwrite files manually, when you upload using the UI, OctoPrint does some metadata storage on the file. You've changed the file contents to something else and OctoPrint has no idea you've done that. To manually upload, put files in the watched folder, don't just manually overwrite files in the uploaded directory.

diesphink, 2018-09-13:
Yeah, didn't thought about that, that perhaps should be the way.

But as I like to keep things organized, and rsyncing with folder structure beats having everything on root dir, I use the upload folder on my deploy setup.

Maybe after #2451 I can go back to using watched folder, but for now I'll just close this, as there can be no bug if I'm not playing nice with the server, :P

closed 2018-09-13 by diesphink