OctoPrint/OctoPrint #2817: Installation error due to sarge 0.1.4 package

redtide, 2018-09-10:
Dispite a recent fix, it seems it doesn't work for me and I get the following error:
```
Collecting sarge==0.1.4 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/bc/9a/a715357f0f0a394edf964fb52b8b39551b629f4d1c996ade5f600fb3d278/sarge-0.1.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-d6twi9_5/sarge/setup.py", line 11, in <module>
        import sarge
      File "/tmp/pip-install-d6twi9_5/sarge/sarge/__init__.py", line 604
        def run(self, input=None, async=False):
                                      ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-d6twi9_5/sarge/
```
This after cloning the repo yesterday following the guide in the [README](https://github.com/foosel/OctoPrint#installation).

ntoff, 2018-09-11:
Add yourself to that list of newbies then because  you've left out a bunch of vital information. The issue template is there for a reason, if you want your issue looked into then please fill it out in its entirety.

jubaleth, 2018-09-11:
Morning, 

I'm stepping in because this issue is taking a counterproductive turn.

@redtide the template is there for a reason, it aids in debugging a problem, and ensures that all information is available. I realize that it takes some time to fill out, but it is certainly a requirement.

Please feel free to re-open this issue with the required information, and I ask that things be kept civil.

Cheers

closed 2018-09-11 by jubaleth

redtide, 2018-09-11:
> Add yourself to that list of newbies then because you've left out a bunch of vital information.
I made reports for various organizations and developers and I never found something like this before, and this is just offending, it's clear that the problem is internal to the dependancy to that package.