OctoPrint/OctoPrint #2819: Feature Request: Use estimated and remaining time from gcode generated by Slic3r Prusa Edition

WilkoV, 2018-09-10:
Time estimation is still a bit tricky in OctoPrint and got pretty reasonable in Slic3r Prusa Edition. And the data is available in the gcode file. So the question is, is it possible to grep the data if the gcode contains:

a) ; generated by Slic3r Prusa Edition
b) the version is 1.41 or better

Data for the estimated time is stored like this and for me it would be good enough to grep the data for normal mode:

; estimated printing time (normal mode) = 8m 39s
; estimated printing time (silent mode) = 8m 39s

And I guess the remaining time in minutes is stored like:

M73 P35 R6

Regards 

Wilko

foosel, 2018-09-14:
You should check out the [PrintTimeGenius Plugin](https://plugins.octoprint.org/plugins/PrintTimeGenius/)? Parsing comments from GCODE files is pretty much a huge part of what it does and AFAIK it also does support the Prusa version of Slic3r. If not, you might want to open a feature request there. I'm not going to add this to core OctoPrint since it adds too much maintenance overhead (new slicer or new slicer version - code adjustment to fix parsing. I already have to play constant catch up with firmwares, I'd rather not open a secondary front of battle)

closed 2018-09-14 by foosel

WilkoV, 2018-09-14:
That's fair. Thx for the hint with the plugin. I'll check it later.