OctoPrint/OctoPrint #2820: Print from Toshiba FlashAir with subfolders doesnt work

2bedom, 2018-09-11:
#### What were you doing?
hi there,

i tried to setup my ender 2 with a Toshiba FlashAir(W-04). 
I'am able to sent files via wifi to the card and select/print them from the Printer Display.
I also see the files in Octoprint BUT I'am not able to select or print them..
This only happens if the file is in a subfolder(i need to specify a subfolder for teh webdav mount on the flashair card, didnt find a possibility to make root available as webdav share)

Scenario:
i places 2 files on the card 
- one in the share(subfolder on card - named ->sliced) a.gcode
- one in the root directory b.gcode
then i called M20
`
Send: M20
Recv: Begin file list
Recv: /SLICED/A~1.GCO 621968
Recv: B7250~1.GCO 621968
Recv: End file list
Recv: ok
`

when i try to select the b.gcode in root , everything works fine
`
Send: M23 /b7250~1.gco
Recv: echo:Now doing file: /b7250~1.gco
Recv: File opened: b7250~1.gco Size: 621968
Recv: File selected
Recv: ok
`
When i try to select the a.gcode i got an error
`
Send: M33 /SLICED/A~1.GCO
Recv: /???/a.gcode
Recv: ok
`
so maybe its a firmware problem, but it only occurs if i try to print from SD-Card with subfolders via octoprint... printing via the printer itself(printer display) works fine.

#### Version of OctoPrint
OctoPrint 1.3.9 auf OctoPi 0.14.0

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Ender 2 
RAMPS mks gen 1.4 
marlin 1.1.9
LCD12864 full Graphic smart

kukulo2011, 2019-04-05:
M23 sliced/a~1.gco   without the slash works, whereas the M23 /sliced/a~1.gco not. This is then issue of the latest Marlin and TH3D firmware?
How to remove the prepending slash from the directory in the M23 command?

kukulo2011, 2019-04-07:
Looks like a problem of parsing the initial / character with firmware of the printer.

I changed in `src/octoprint/util/comm.py` : 

`if sd:
                        if not self.isOperational():
                                # printer is not connected, can't use SD
                                return

                        if filename.startswith("/") and self._sdRelativePath:
                                filename = filename[1:]

                        self._sdFileToSelect = filename
                        self._sdFileToSelectUser = user
                        self.sendCommand("M23 %s" % filename, tags=tags | {"trigger:comm.select_file",})
                else:
                        self._currentFile = PrintingGcodeFileInformation(filename,
                                                                         offsets_callback=self.getOffsets,
                                                                         current_tool_callback=self.getCurrentTool,
                                                                         user=user)
                        self._callback.on_comm_file_selected(filename, self._currentFile.getFilesize(), False, use$

        def unselectFile(self):
                if self.isBusy():
                        return
`

to 

`if sd:
                        if not self.isOperational():
                                # printer is not connected, can't use SD
                                return

                        if filename.startswith("/") and self._sdRelativePath:
                                filename = filename[1:]

                        filename = filename.lstrip("/")
                        self._sdFileToSelect = filename
                        self._sdFileToSelectUser = user
                        self.sendCommand("M23 %s" % filename, tags=tags | {"trigger:comm.select_file",})
                else:
                        self._currentFile = PrintingGcodeFileInformation(filename,
                                                                         offsets_callback=self.getOffsets,
                                                                         current_tool_callback=self.getCurrentTool,
                                                                         user=user)
                        self._callback.on_comm_file_selected(filename, self._currentFile.getFilesize(), False, use$

        def unselectFile(self):
                if self.isBusy():
                        return
`

Stop octoprint, then reinstall the octoprint within venv with:
`./venv/bin/python setup.py install`
Start octoprint.

After this the problem is solved.

foosel, 2019-04-08:
Try if

[image]

fixes it.

kukulo2011, 2019-04-08:
This works as well. 
Somehow the firmware auto detection does not detect this.
On stock Ender 3 the absolute path was working, on latest TH3D on Ender 3 not.

foosel, 2019-04-08:
Then something got broken in that firmware fork. Auto detection doesn't hold her because this setting is usually only need for RepRapFirmware, not Marlin derivatives.

kukulo2011, 2019-04-08:
Strange I used the TH3D_UFW_U1.R2.9b firmware to flash the printer and this is Marlin based. I checked the source code of the firmware for the file opening from SD card and it is the same as in the Marlin firmware. Also tried screen to send directly commands to the printer and the slash in the beginning is not accepted for directories, only for the files in SD card root.

foosel, 2019-04-11:
Maybe something changed in the default Marlin behaviour then as well and you are the first to run into this.

Could you please provide a full `serial.log` of initial connect to your printer that should also include a response to `M20`?

kukulo2011, 2019-04-26:
`2019-04-26 19:26:22,595 - Enabling serial logging
2019-04-26 19:26:23,252 - Send: M105
2019-04-26 19:26:23,336 - Recv: ok T:52.50 /0.00 B:46.50 /0.00 @:0 B@:0
2019-04-26 19:26:27,823 - Connection closed, closing down monitor
2019-04-26 19:26:27,841 - Changing monitoring state from "Operational" to "Offline"
2019-04-26 19:26:30,718 - Changing monitoring state from "Offline" to "Detecting serial port"
2019-04-26 19:26:30,773 - Serial port list: ['/dev/ttyUSB0']
2019-04-26 19:26:30,776 - Connecting to: /dev/ttyUSB0
2019-04-26 19:26:30,844 - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2019-04-26 19:26:30,851 - Connected to: Serial<id=0xaa8f1b30, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2019-04-26 19:26:30,852 - Starting baud rate detection...
2019-04-26 19:26:30,853 - Changing monitoring state from "Opening serial port" to "Detecting baudrate"
2019-04-26 19:26:31,866 - Trying baudrate: 115200
2019-04-26 19:26:31,873 - Send: N0 M110 N0*125
2019-04-26 19:26:32,131 - Recv: start
2019-04-26 19:26:32,138 - Changing monitoring state from "Detecting baudrate" to "Operational"
2019-04-26 19:26:32,181 - Send: N0 M110 N0*125
2019-04-26 19:26:32,184 - Recv: echo:Marlin TH3D U1.R2.9b
2019-04-26 19:26:32,192 - Recv: 
2019-04-26 19:26:32,202 - Recv: echo: Last Updated:  | Author: (TH3D)
2019-04-26 19:26:32,220 - Recv: echo:Compiled: Apr  5 2019
2019-04-26 19:26:32,235 - Recv: echo: Free Memory: 11513  PlannerBufferBytes: 1488
2019-04-26 19:26:32,239 - Recv: echo:EEPROM version mismatch (EEPROM=? Marlin=V55)
2019-04-26 19:26:32,241 - Recv: echo:Hardcoded Default Settings Loaded
2019-04-26 19:26:36,781 - Recv: echo:SD card ok
2019-04-26 19:26:36,794 - Recv: ok
2019-04-26 19:26:36,798 - Send: M115
2019-04-26 19:26:36,834 - Recv: FIRMWARE_NAME:Marlin TH3D U1.R2.9b SOURCE_CODE_URL: PROTOCOL_VERSION:1.0 MACHINE_TYPE:TH3D U1.R2.9b EXTRUDER_COUNT:1 UUID:
2019-04-26 19:26:36,838 - Recv: ok
2019-04-26 19:26:36,850 - Send: M21
2019-04-26 19:26:36,882 - Recv: echo:SD card ok
2019-04-26 19:26:36,885 - Recv: ok
2019-04-26 19:26:36,890 - Send: M117 IP 192.168.X.X
2019-04-26 19:26:36,914 - Recv: ok
2019-04-26 19:26:36,919 - Send: M20
2019-04-26 19:26:36,946 - Recv: Begin file list
2019-04-26 19:26:37,122 - Recv: /47ACB~1.MOD/TEST/TEST-D~1.GCO 16902701
2019-04-26 19:26:37,139 - Recv: TEST-D~1.GCO 16902719
2019-04-26 19:26:37,197 - Recv: /ENDER3~2/CE3_CH~1.GCO 4861442
2019-04-26 19:26:37,201 - Recv: /ENDER3~2/CE3_CO~1.GCO 1399517
2019-04-26 19:26:37,204 - Recv: /ENDER3~2/EXTRUD~1.GCO 1278954
2019-04-26 19:26:37,211 - Recv: /ENDER3~2/EXTRUD~2.GCO 2083068
2019-04-26 19:26:37,214 - Recv: /ENDER3~2/FAN_CO~1.GCO 2687163
2019-04-26 19:26:37,217 - Recv: /ENDER3~2/LCD_BO~1.GCO 3337313
2019-04-26 19:26:37,221 - Recv: /ENDER3~2/LINK_V~1.GCO 658225
2019-04-26 19:26:37,227 - Recv: /ENDER3~2/CE3_CA~1.GCO 838897
2019-04-26 19:26:37,238 - Recv: /ENDER3~2/CE3_CA~2.GCO 1894269
2019-04-26 19:26:37,240 - Recv: /ENDER3~2/CE3_CA~3.GCO 1539491
2019-04-26 19:26:37,243 - Recv: /ENDER3~2/CE3_LI~1.GCO 2256484
2019-04-26 19:26:37,245 - Recv: /ENDER3~2/CE3_40~1.GCO 1468057
2019-04-26 19:26:37,247 - Recv: /ENDER3~2/CE3_BE~1.GCO 1183864
2019-04-26 19:26:37,255 - Recv: CE3_TE~1.GCO 1340502
2019-04-26 19:26:37,260 - Recv: /TOMITES/CE3_TO~1.GCO 559675
2019-04-26 19:26:37,264 - Recv: /TOMITES/CE3_TO~2.GCO 476770
2019-04-26 19:26:37,268 - Recv: XYZCUB~1.GCO 507807
2019-04-26 19:26:37,271 - Recv: /THERMO~1/THERMO~1.GCO 26671149
2019-04-26 19:26:37,274 - Recv: /THERMO~1/THERMO~2.GCO 3169856
2019-04-26 19:26:37,278 - Recv: /CSIPESZ/CE3_CS~1.GCO 333951
2019-04-26 19:26:37,280 - Recv: /CSIPESZ/CE3_CS~2.GCO 341986
2019-04-26 19:26:37,284 - Recv: TUX~1.GCO 4016299
2019-04-26 19:26:37,297 - Recv: /RPC/CE3_CO~1.GCO 2468153
2019-04-26 19:26:37,301 - Recv: /RPC/CE3_UN~1.GCO 6219002
2019-04-26 19:26:37,306 - Recv: /RPC/COVER~1.GCO 1819693
2019-04-26 19:26:37,311 - Recv: /RPC/CE3_UN~2.GCO 6931222
2019-04-26 19:26:37,313 - Recv: /RPC/CE3_CO~2.GCO 2371536
2019-04-26 19:26:37,317 - Recv: /RPC/CE3_DE~1.GCO 2562256
2019-04-26 19:26:37,320 - Recv: CE3_CA~1.GCO 443454
2019-04-26 19:26:37,349 - Recv: /TRASH-~1/FILES/SLUCHA~1.GCO 532546
2019-04-26 19:26:37,352 - Recv: /TRASH-~1/FILES/CE3_PP~1.GCO 1662750
2019-04-26 19:26:37,355 - Recv: /TRASH-~1/FILES/CE3_EX~1.GCO 2081679
[... 77 lines cut ...]
2019-04-26 19:26:38,649 - Recv: /TPE/CE3_TO~1.GCO 476197
2019-04-26 19:26:38,697 - Recv: /FAN/FANOFF~1.GCO 192
2019-04-26 19:26:38,700 - Recv: /FAN/FANON~1.GCO 192
2019-04-26 19:26:38,703 - Recv: /FAN/02MM~1.GCO 50
2019-04-26 19:26:38,706 - Recv: /FAN/FANOFF~3.GCO 8
2019-04-26 19:26:38,710 - Recv: /FAN/FANOFF~4.GCO 8
2019-04-26 19:26:38,715 - Recv: /FAN/FANOFF~5.GCO 8
2019-04-26 19:26:38,718 - Recv: /FAN/FANOFF~6.GCO 8
2019-04-26 19:26:38,722 - Recv: /EXTRUD~1/CE3_EN~1.GCO 332696
2019-04-26 19:26:38,729 - Recv: /EXTRUD~1/CE3_SP~1.GCO 402090
2019-04-26 19:26:38,744 - Recv: /SPOOLH~1/CE3_SP~1.GCO 10675623
2019-04-26 19:26:38,762 - Recv: CE3_PE~1.GCO 371758
2019-04-26 19:26:38,798 - Recv: /BED_LE~1/CE3_BE~1.GCO 32810
2019-04-26 19:26:38,801 - Recv: /BED_LE~1/SQ_BED~1.GCO 57947
2019-04-26 19:26:38,804 - Recv: /BED_LE~1/CE3_FD~1.GCO 71088
2019-04-26 19:26:38,844 - Recv: /FLEXIB~1/CE3_PP~1.GCO 577822
2019-04-26 19:26:38,846 - Recv: /FLEXIB~1/CE3_EX~2.GCO 2791018
2019-04-26 19:26:38,848 - Recv: /FLEXIB~1/CE3_TP~1.GCO 577919
2019-04-26 19:26:38,851 - Recv: /FLEXIB~1/CE3_EX~1.GCO 2104031
2019-04-26 19:26:38,874 - Recv: /PP/CE3_CO~1.GCO 2503752
2019-04-26 19:26:38,879 - Recv: /PP/CE3_PP~1.GCO 2546868
2019-04-26 19:26:38,882 - Recv: /PP/CE3_PP~2.GCO 910743
2019-04-26 19:26:38,885 - Recv: /PP/CE3_PP~3.GCO 1726643
2019-04-26 19:26:38,889 - Recv: End file list
2019-04-26 19:26:38,893 - Recv: ok`

kukulo2011, 2019-04-26:
Above is the serial.log.

foosel, 2019-05-13:
So your firmware reports the file path with a leading slash but then refuses to open it with that exact filename with the leading slash. That's a firmware bug. It needs to either support the leading slash for opening, or not put it in the `M20` output. 

Pushing a filename format to a client that then isn't supported for actual file management doesn't make a lot of sense and is probably unintentional.

dcarusil, 2019-07-12:
Just wanted to say thanks! An Ender 3 owner trying to work around usb serial timeout issues by running off the SD through octopi/print. The printer firmware is the community choice.. months later and I guess nobody over at TH3D knows or cares?

FIRMWARE_NAME:Marlin TH3D U1.R2.11 SOURCE_CODE_URL: PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 

Anyway thanks a bunch!

condatek, 2019-10-07:
I solved the exacts same problem but with an Ender 3 Pro flashed with Marlin 1.1.9 Bugfix. Octoprint was able to read contents of SD card but no file in subfolders could be opened. Now it is working properly, thanks!