OctoPrint/OctoPrint #2822: [Question/Request] Being able to operate a cooling fan at a set CPU temp for the Raspberry Pi 3 B+

Bobertsawesome, 2018-09-14:
Hi there, I have never dealt with Raspbian ecosystems or Python for that matter.

I have built a circuit according to the guide, and I am wanting to run a script for the Raspberry Pi using this tutorial: https://www.instructables.com/id/Control-a-Cooling-Fan-on-a-Raspberry-Pi-3/

Since octoprint runs on its own OS, I am not sure how to get this working with octoprint on startup, etc. What exactly needs to be done to make this work on octoprint?

foosel, 2018-09-14:
For questions or other kind of help or support, please get in touch on the [community forum](https://discourse.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2018-09-14 by foosel

Bobertsawesome, 2018-09-14:
Thanks for letting me know. I will delete the thread.