OctoPrint/OctoPrint #2823: [BUG] Timeout settings not being saved

bolsoncerrado, 2018-09-14:
#### What were you doing?

Altering Octoprint Timeout Settings

#### What did you expect to happen?

Octoprint to SAVE the changes I made

#### What happened instead?

Octoprint did NOT save the changes

#### Did the same happen when running OctoPrint in safe mode?

Didnt test yet.

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

OctoPi 0.15.0

#### Printer model & used firmware incl. version

Prusa MK2.5 3.3.1

#### Browser and version of browser, operating system running browser

Chrome 69
Safari 11
Firefox 62

#### Link to octoprint.log

Unrelated

#### Link to contents of terminal tab or serial.log

Unrelated

#### Screenshot(s)/video(s) showing the problem:

I can record a screencap vid if required?

I have read the FAQ.

foosel, 2018-09-14:
Duplicate of #2780 

> Link to octoprint.log
>
> Unrelated

@bolsoncerrado, how often do I have to repeat to you that I *NEED* log files in order to be able to analyse issues? If that issue hadn't by chance been something that was already reported I would have ABSOLUTELY NOTHING to go on now.

closed 2018-09-14 by foosel

foosel, 2018-09-16:
`raspi.log`?! The ticket template clearly states to always include `octoprint.log`.

bolsoncerrado, 2018-09-17:
I meant the octoprint.log yes, sorry.