OctoPrint/OctoPrint #2824: [Safari] Bug when opening setting while in control tab.

Moorviper, 2018-09-14:
Settings doesn't appear when open it while the control tab

Client:
OS: High Sierra  Safari Version 11.1.2 (13605.3.8)
https://twitter.com/moorviper/status/1040670946183536647

Server
Ubuntu Mate : Octoprint 1.3.9

Moorviper, 2018-09-19:
same is habbening when downloading something from thingiverse / i will compare the code on the weekend ...

foosel, 2018-09-20:
Can you please also provide all the info requested in the ticket template as I requested on Twitter? ;) Because right now this issue isn't really actionable.

Moorviper, 2018-09-21:
With Safari Version 12.0 (13606.2.11) it is fixed.
Can't reproduce it anymore :-/
I had activated automatic updates ...

closed 2018-09-21 by Moorviper

foosel, 2018-09-21:
Less work for me ;) Hope the exam went well!

Moorviper, 2018-09-21:
THX
There was the same idct implementation exam task as last year :D
This year it seamed to work out much better, but i know it when i get the results....
Now i have 4 weeks time to work on octofarm 

[image]

This time i write the documentation when i do the installation ^^

foosel, 2018-09-21:
Oh, sweet, LaTeX :D It's been ages since I last used that, but I still have my bible aka "The LaTeX Compendium" ;)

And also 👍 for the project :D

Moorviper, 2018-09-21:
latex is great in most of the times / i am using luaLaTeX with arara so i can configure the build process on the top on the tex main file

I use it in combination with travis.ci (payed acc for free via github education as long as student)

https://github.com/Moorviper/Freifunk-Router-Anleitungen

When i tag the release round about 80 pdf are dynamically are created from text blocks (one for each router)
It is still in progress.
But the motivation dropped as the community in Fulda began to get dump.
So i do no Freifunk anymore because i don't want to use the Frankfurt firmware in Fulda :D
That doesn't make sense ...

foosel, 2018-09-21:
> Is it a good idea to write a octoprint plugin which send the apikey to a central instance as Octofarm to auto register the octoprint instance ?

Risky, since it could easily be abused. I'm currently working on a new feature for OctoPrint 1.3.10 though that will include a new authentication work flow ("Application Foobar would like to control your OctoPrint instance, accept or deny?") that might allow you something like pasting an URL to a (local) OctoPrint instance into your fram interface (or even auto detect them via MDNS or UPNP) and then trigger that authentication workflow.