OctoPrint/OctoPrint #2831: Error:No Checksum with line number, Last Line:

gregvis, 2018-09-27:
Getting this repeating message in the terminal. Print has stopped in the middle of printing 

Send: N13312 G1 X100.16 Y136.459\x00\x00??\x04P??\x04\x00S?\x04 ?????\x1a\x01???\x04\x00Z?\x04???\x04?????\x1a\x00?\x0f?\x00*102
Recv: Error:No Checksum with line number, Last Line: 13311
Recv: Resend: 13312
Recv: ok
Send: N13312 G1 X100.16 Y136.459\x00\x00??\x04P??\x04\x00S?\x04 ?????\x1a\x01???\x04\x00Z?\x04???\x04?????\x1a\x00?\x0f?\x00*102
Recv: Error:No Checksum with line number, Last Line: 13311
Recv: Resend: 13312
Recv: ok

#### What were you doing?

Ideally provide exact steps to follow in order to reproduce your problem:

1. Slice with Cura 3.4.1
2. Print with Octoprint (using Cura plugin)
3. Print starts
4. Print stops at ~50% complete.

#### What did you expect to happen?
Print to finish
#### What happened instead?
Stopped mid print
#### Did the same happen when running OctoPrint in safe mode?

Didn't try

#### Version of OctoPrint 
1.3.9

#### Operating System running OctoPrint 
Octopi

#### Printer model & used firmware incl. version
Anet A8 - Marlin 1.19

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-09-27:
Logs please.

gregvis, 2018-09-29:
[octoprint.log](https://drive.google.com/open?id=1OSsDW2WKutyoZjAeYYN3E6Pd5CobRJzW
)

foosel, 2018-09-29:
I need a serial.log and I also need you to check whether the issue persists in safe mode. Just fill in the blanks in the ticket template that you so far ignored please, all that stuff is asked for for a reason.

closed 2018-10-19 by foosel