OctoPrint/OctoPrint #2832: OP does not automatically connect to printer at start-up

hoegge, 2018-10-02:
#### What were you doing?
Starting Octoprint (booting from shutdown of RPi or restart), the "Auto-connect on server startup" checkbox is already set before shutdown both in front page and settings.

The log seems to indicate it has connected to the printer but the GUI does not reflect that.

1. Starter server
2. Check if printer is connected via serial 
3. It is not

#### What did you expect to happen?
Octoprint to connect to printer

#### What happened instead?
It did not connect

#### Did the same happen when running OctoPrint in safe mode?
Yes - no difference in safe mode.

#### Version of OctoPrint
1.3.9 - clean new install of 1.3.8 and updated to 1.3.9. Only plug-in added by me is automatic shutdown

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
CR-10 stock firmware

#### Browser and version of browser, operating system running browser
Chrome, Win10

#### Link to octoprint.log
https://www.dropbox.com/s/c83grjkhn2sepc6/octoprint.log?dl=0

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
https://www.dropbox.com/s/8jnr2rud7lr30k2/js%20console.txt?dl=0

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-10-19:
```
2018-09-30 16:39:31,059 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
[...]
2018-09-30 16:39:31,092 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2018-09-30 16:39:31,096 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2018-09-30 16:39:31,101 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
[...]
2018-09-30 16:40:04,805 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Offline"
```

This looks like it is in fact trying to connect during startup, but then failing to handshake with the printer for some reason. I'll need a full `serial.log` to diagnose this further. 

Please enable `serial.log` in Settings > Serial Connection, then restart OctoPrint and provide the resulting log.

I also notice that you have it detect the serial port instead of setting that to a fixed value. Try if selecting the proper serial port instead of "AUTO" for the connection settings changes the behaviour.

foosel, 2018-11-15:
Closing due to lack of response.

closed 2018-11-15 by foosel

hoegge, 2019-02-24:
Sorry for the (too) late reply but I had just moved and my printers were not set up. The serial log looks pretty normal, as if it does not try to connect until I had pushed the connect button:
``` 
2019-02-24 08:21:05,171 - Changing monitoring state from "Offline" to "Detecting serial port"
2019-02-24 08:21:05,202 - Serial port list: ['/dev/ttyUSB0']
2019-02-24 08:21:05,203 - Connecting to: /dev/ttyUSB0
2019-02-24 08:21:05,210 - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2019-02-24 08:21:05,213 - Connected to: Serial<id=0x68017ed0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2019-02-24 08:21:05,214 - Changing monitoring state from "Opening serial port" to "Connecting"
2019-02-24 08:21:05,224 - Send: N0 M110 N0*125
2019-02-24 08:21:05,452 - Recv: start
2019-02-24 08:21:05,455 - Recv: echo: External Reset
2019-02-24 08:21:05,456 - Send: N0 M110 N0*125
2019-02-24 08:21:05,457 - Recv: Marlin 1.0.0
2019-02-24 08:21:05,458 - Recv: echo: Last Updated: Oct 24 2016 12:00:03 | Author: (none, default config)
2019-02-24 08:21:05,460 - Recv: Compiled: Oct 24 2016
2019-02-24 08:21:05,461 - Recv: echo: Free Memory: 11431  PlannerBufferBytes: 1232
2019-02-24 08:21:05,462 - Recv: echo:Hardcoded Default Settings Loaded
2019-02-24 08:21:05,463 - Recv: echo:Steps per unit:
2019-02-24 08:21:05,465 - Recv: echo:  M92 X80.00 Y80.00 Z400.00 E93.00
2019-02-24 08:21:05,466 - Recv: echo:Maximum feedrates (mm/s):
2019-02-24 08:21:05,467 - Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
2019-02-24 08:21:05,468 - Recv: echo:Maximum Acceleration (mm/s2):
2019-02-24 08:21:05,470 - Recv: echo:  M201 X500 Y500 Z100 E5000
2019-02-24 08:21:05,471 - Recv: echo:Acceleration: S=acceleration, T=retract acceleration
2019-02-24 08:21:05,493 - Recv: echo:  M204 S500.00 T500.00
2019-02-24 08:21:05,494 - Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
2019-02-24 08:21:05,495 - Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
2019-02-24 08:21:05,496 - Recv: echo:Home offset (mm):
2019-02-24 08:21:05,498 - Recv: echo:  M206 X0.00 Y0.00 Z0.00
2019-02-24 08:21:05,499 - Recv: echo:PID settings:
2019-02-24 08:21:05,500 - Recv: echo:   M301 P21.73 I1.54 D76.55
2019-02-24 08:21:07,156 - Recv: echo:SD card ok
2019-02-24 08:21:07,172 - Recv: ok
2019-02-24 08:21:07,178 - Changing monitoring state from "Connecting" to "Operational"
2019-02-24 08:21:07,201 - Send: N0 M110 N0*125
2019-02-24 08:21:07,254 - Recv: ok
2019-02-24 08:21:07,257 - Send: N1 M115*39
2019-02-24 08:21:07,322 - Recv: FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://www.mendel-parts.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
2019-02-24 08:21:07,342 - Recv: ok
2019-02-24 08:21:07,365 - Send: M20
2019-02-24 08:21:07,407 - Recv: Begin file list
2019-02-24 08:21:07,411 - Recv: GO_Z03.GCO
2019-02-24 08:21:07,415 - Recv: MHPLEV~1.GCO
2019-02-24 08:21:07,420 - Recv: PLA_HE~1.GCO
2019-02-24 08:21:07,426 - Recv: RISTLO~1.GCO
2019-02-24 08:21:07,428 - Recv: End file list
2019-02-24 08:21:07,431 - Recv: ok
2019-02-24 08:21:12,180 - Send: M105
2019-02-24 08:21:12,199 - Recv: ok T:18.7 /0.0 B:20.5 /0.0 T0:18.7 /0.0 @:0 B@:0
2019-02-24 08:21:15,280 - Send: M104 S200
2019-02-24 08:21:15,295 - Recv: ok
2019-02-24 08:21:17,184 - Send: M105
2019-02-24 08:21:17,203 - Recv: ok T:18.7 /200.0 B:20.5 /0.0 T0:18.7 /200.0 @:127 B@:0
2019-02-24 08:21:17,896 - Send: M140 S60
2019-02-24 08:21:17,969 - Recv: ok
2019-02-24 08:21:19,187 - Send: M105
2019-02-24 08:21:19,200 - Recv: ok T:19.4 /200.0 B:20.5 /60.0 T0:19.4 /200.0 @:127 B@:0
2019-02-24 08:21:21,189 - Send: M105
2019-02-24 08:21:21,203 - Recv: ok T:20.2 /200.0 B:20.5 /60.0 T0:20.2 /200.0 @:127 B@:127
...
```

seltix5, 2019-08-26:
Hello,
I have the same problem. I found this issue https://github.com/foosel/OctoPrint/issues/2311 which apparently describes the same problem but it's closed too. Is there already a new issue created for this bug?