OctoPrint/OctoPrint #2833: Too many consecutive timeouts

purist69, 2018-10-03:
This issue happens on two separate board, a Raspberry Pi Model 3B+ with the 0.15.1 image installed and a custom install on a Rock64 with DietPi.  Both work fine with my i3 clone.

#### What were you doing?

1. Load web interface
2. Click 'Connect'
3. Wait 2-3 minutes

#### What did you expect to happen?

Successfully connect without error

#### What happened instead?

The following error popped up:

Communication error
There was a communication error while talking to your printer. Please consult the terminal output and octoprint.log for details. Error: Too many consecutive timeouts, printer still connected and alive?

#### Did the same happen when running OctoPrint in safe mode?

yes

#### Version of OctoPrint

1.39

#### Operating System running OctoPrint

official octopi image 0.15.1 running on a RPi 3B+ AND
DietPi 6.16 on Rock64

#### Printer model & used firmware incl. version

Prusa i3 MK3 with firmware 3.4.0-1327

#### Browser and version of browser, operating system running browser

Chrome 69 on Windows 10

#### Link to octoprint.log

2018-10-03 02:31:23,558 - octoprint.startup - INFO - ******************************************************************************
2018-10-03 02:31:23,560 - octoprint.startup - INFO - Starting OctoPrint 1.3.9
2018-10-03 02:31:23,561 - octoprint.startup - INFO - ******************************************************************************
2018-10-03 02:31:23,631 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2018-10-03 02:31:23,633 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-10-03 02:31:25,463 - octoprint.plugin.core - INFO - Found 21 plugin(s) providing 21 mixin implementations, 19 hook handlers
2018-10-03 02:31:25,560 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-10-03 02:31:25,587 - octoprint.server - INFO - Intermediary server started
2018-10-03 02:31:25,588 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-10-03 02:31:26,193 - octoprint.plugin.core - INFO - Found 21 plugin(s) providing 21 mixin implementations, 19 hook handlers
2018-10-03 02:31:26,230 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-10-03 02:31:26,879 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-10-03 02:31:27,092 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-10-03 02:31:28,928 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-10-03 02:31:30,757 - octoprint.util.pip - INFO - Version of pip is 9.0.3
2018-10-03 02:31:30,759 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-10-03 02:31:30,760 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-10-03 02:31:30,765 - octoprint.plugin.core - INFO - Initialized 21 plugin implementation(s)
2018-10-03 02:31:30,776 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Display ETA (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eta
|  DisplayZ (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayz
|  FileManager (0.1.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filemanager
|  Fullscreen Plugin (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
[... 68 lines cut ...]
2018-10-03 02:35:44,232 - Connected to: Serial<id=0x6db69350, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-10-03 02:35:44,232 - Starting baud rate detection...
2018-10-03 02:35:44,233 - Changing monitoring state from "Opening serial port" to "Detecting baudrate"
2018-10-03 02:35:45,238 - Trying baudrate: 115200
2018-10-03 02:35:45,244 - Send: N0 M110 N0*125
2018-10-03 02:35:45,353 - Recv: start
2018-10-03 02:35:45,355 - Changing monitoring state from "Detecting baudrate" to "Operational"
2018-10-03 02:35:45,368 - Send: N0 M110 N0*125
2018-10-03 02:36:15,395 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-10-03 02:36:15,404 - Send: N1 M115*39
2018-10-03 02:36:45,434 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-10-03 02:36:45,444 - Send: N2 M21*18
2018-10-03 02:37:15,473 - No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-10-03 02:37:15,485 - Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2018-10-03 02:37:15,495 - Connection closed, closing down monitor

#### Link to contents of Javascript console in the browser

I dont know how to get this

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

bolsoncerrado, 2018-10-15:
https://github.com/foosel/OctoPrint/issues/2843#issuecomment-429759157 ?

foosel, 2018-10-15:
Indeed looks like a duplicate of #2843 - and a firmware bug, not OctoPrint.

closed 2018-10-15 by foosel