OctoPrint/OctoPrint #2836: I have an issue with octoprint that needs the attention of an expert. octoprint.log error report

Cheezeh3d, 2018-10-05:
#### What were you doing?

1. I tried to print a gcode file and I received an error that told me to look at the octoprint.log file to find the error. 
2. After spending a few minutes, I found that octoprint.util.comm reported sever errors that stated "Caught an exception in the send loop". It repeated this error a few times until it reported "ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker"

#### What did you expect to happen?
I expected the printer to start printing my file
#### What happened instead?
An error was reported and it redirected me to the octoprint.log file where I found several errors that reported something creashing inside the serial connection loop. 
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
OctoPrint 1.3.9

#### Operating System running OctoPrint
MacOS

#### Printer model & used firmware incl. version
CR-10 S and IDK the firmware

#### Browser and version of browser, operating system running browser
Chrome Version 69.0.3497.100 (Official Build) (64-bit)

#### Link to octoprint.log
https://gist.github.com/Cheezeh3d/10a8bcdf2d426f9a0272cf084b4fbfbb

#### Link to contents of terminal tab or serial.log
unsure

#### Link to contents of Javascript console in the browser
unsure

#### Screenshot(s)/video(s) showing the problem:
https://drive.google.com/file/d/1Yqu8ct1onTnvMD8F323xpyhuF9npQHJ_/view?usp=sharing

I have read the FAQ.

razerraz, 2018-10-09:
well, I'm not expert, but I can see two things : 
internet access problem (proxy ?), but it's not terminal
some unicode to ascii conversion failing that I presume OS related (macOS)

please enable debug logging in order to see what kind of unicode strings causing that.

Cheezeh3d, 2018-10-09:
> well, I'm not expert, but I can see two things :
> internet access problem (proxy ?), but it's not terminal
> some unicode to ascii conversion failing that I presume OS related (macOS)
> 
> please enable debug logging in order to see what kind of unicode strings causing that.

Cheezeh3d, 2018-10-09:
Thanks for the reply razerraz. I'll look more into the debug log.

foosel, 2018-10-15:
Please provide a file with which to reproduce this.

Cheezeh3d, 2018-10-15:
Actually,

Cheezeh3d, 2018-10-15:
Actually, I figured it out. It was an issue with the GCode file. I don't know what the issue was and I don't know why it affected the connection to the printer, but I deleted the GCode file and everything worked. Thanks for all the feedback!

closed 2018-10-15 by Cheezeh3d