OctoPrint/OctoPrint #2837: [Request] Live updating of Gcode

wehiird, 2018-10-08:
Description and elaboration in this reddit post:

https://www.reddit.com/r/3Dprinting/comments/9ltjk5/anybody_know_anything_about_live_gcode_updating/

#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

wehiird, 2018-10-18:
> Could you please summarize the important bits from the reddit thread that
> make up this request?

"Anybody know anything about live gcode updating via any slicer?"

"I assume you're referring to the ability to tune settings mid-print? I'm
not aware of any; it's more common to print things like temperature towers,
with different settings on different sections of the model which can be
examined after the print is done. The aforementioned temperature towers are
the most common, but you can also adapt the method for just about
anything—retraction, z-height, extrusion multiplier and so on."

"right, but say i am printing six things, and one of them starts to peel up
off the bed. so, i it might be easier to just eliminate one of the objects
and let the others continue printing..

does that make sense?"

There's more, but does that help at all?

truglodite, 2018-10-18:
You should look in to using the exclude region plugin. I use it for that exact thing... skipping failed parts on a multipart plate.

foosel, 2018-10-18:
Definitely sounds like there indeed already is a [plugin for that usecase](https://plugins.octoprint.org/plugins/excluderegion/) as mentioned. And just editing a file while it is being printed is something that is just asking for trouble - how do you know which parts are still possible to allow to edit to the user considering the time constraints involved? And how do you prevent the estimator from being thrown off completely? 

If the above plugin doesn't already cover the use case anyhow, this is a strong contender not for core OctoPrint but someone writing a plugin to do it, possible with some adjustments in OctoPrint to make that possible (even though I can't think of anything that would be needed right now).

wehiird, 2018-10-19:
Thank you for responding. A colleague and I have been talking about writing
our own slicing software for a while for various reasons, one of which is
this feature. I figured before we went too far down the rabbit hole I would
ask around in some forums to find out if anything like it already exists.
Octoprint was suggested which I haven't ever actually used, but in the
Makerspace I work at we are planning on outfitting our two Taz's and Prusa
with Octoprint so that we can do all of the awesome stuff Octoprint will
allow us to do.

Thanks again.

foosel, 2018-11-15:
> @wehiird can you please verify if the linked plugin already does what's requested here?

Still waiting for an answer on this, @wehiird ;)

DanielJoyce, 2019-02-08:
At the point where you need to make this decision, the gcode has already been generated. So slicer won't change that.

closed 2019-02-12 by foosel