OctoPrint/OctoPrint #2839: comm.regex_position does not allow spaces after a ':' character

FormerLurker, 2018-10-09:
I've been debugging an issue from a user that caused a lockup (due to signal wait) whenever Octolapse requests a position via M114.  After some digging I discovered that the response for M114 from this printer (AlfaWise U20) was not being recognized by the comm.regex_position regex search because there are spaces after the colon.  I have adjusted the regex to account for this:
```
regex_position = re.compile("X:\s*(?P<x>{float})\s*Y:\s*(?P<y>{float})\s*Z:\s*(?P<z>{float})\s*((E:\s*(?P<e>{float}))|(?P<es>(E\d+:{float}\s*)+))".format(float=regex_float_pattern))
```
Here is a sample of the position response from the printer:
```
ok X:150.0 Y:150.0 Z:  0.7 E:  0.0
```
I believe it's using a fixed width format for the parameters here, which is why the original regex doesn't work.  I've tested this pattern, but there could be some situations I didn't take into account (though I can't think of any).  Basically I just added a `\s*` after each colon, but NOT to the es named capture group because there is some stuff going on there that I don't understand.

I hope this helps!

DC43star, 2018-10-09:
Thanks again for the help!

FormerLurker, 2018-10-15:
@foosel, let me know if you need any additional information when you are available.  I don't want the bot to delete this issue, and I believe I've supplied everything necessary, but I could be wrong :)

foosel, 2018-10-16:
@FormerLurker in this case all's fine. Just need to look into it a bit more closely, but on first glance it looks like allowing whitespace there should hopefully not break anything. Just really wish firmware vendors would stop doing their own thing all the time and divert from established defaults 🙄

FormerLurker, 2018-10-17:
Tell me about it!  We need to form a standards committee :)

FYI, I've bypassed the on_event Events.POSITION_UPDATE event and have been using the regex I posted (as well as some modified versions of the comm.py code) within the octoprint.comm.protocol.gcode.received hook to detect position responses myself, and haven't had any issues so far.  I'll let you know immediately if I find any issues with the regex.

foosel, 2018-10-18:
Adjusted by the above commit, already available on `maintenance`, soon `devel` and to be pushed out as part of 1.3.10.

I also took the opportunity to add a unit test for it referring to this ticket so I won't wonder down the road ;)

FormerLurker, 2018-10-18:
Excellent, thank you!  I'll remove my own position update and will go back to using Events.POSITION_UPDATE.  Good idea regarding the unit test.  If I see any other odd position responses, I'll let you know.

Closed!

commit 7b5d2e3e (2018-10-18): Allow whitespace after : in M114 responses

Closes #2839
closed 2018-10-18 by FormerLurker