OctoPrint/OctoPrint #2842: "tornado.access - WARNING - 409 GET /api/printer (127.0.0.1)" issued permanntly

thomaskilian, 2018-10-10:
#### What were you doing?

Just rebooting and octoprint gets and keeps totally busy

#### What did you expect to happen?

Come up and react to web connections (as it did in the past)

#### What happened instead?

octoprint is busy producing

    tornado.access - WARNING - 409 GET /api/printer (127.0.0.1)

in the log file. This causes the app to become unresponsive.

#### Did the same happen when running OctoPrint in safe mode?

I haven't tried since I don't know how to start it in save mode.

#### Version of OctoPrint

Too bad, it does not come that far in the web interface. I had been up to date all time so far.

#### Operating System running OctoPrint

/etc/octopi_version: 0.13.0

#### Printer model & used firmware incl. version

I'm not printing with my Pi at all

#### Browser and version of browser, operating system running browser

Safari 12.0 (13606.2.11)

#### Link to octoprint.log

Well, it's filled with the above mentioned line. Right in front I find

2018-10-10 11:16:43,551 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2018-10-10 11:16:43,750 - octoprint.server - INFO - Intermediary server started
2018-10-10 11:16:43,754 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/
plugins and installed plugin packages...
2018-10-10 11:16:47,082 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2018-10-10 11:16:47,372 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-10-10 11:16:50,899 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-10-10 11:16:51,120 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-10-10 11:17:03,443 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2018-10-10 11:17:15,443 - octoprint.util.pip - INFO - Version of pip is 8.1.1
2018-10-10 11:17:15,462 - octoprint.plugin.core - INFO - Initialized 7 plugin implementation(s)
2018-10-10 11:17:15,513 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/plugins/virtual_printer
2018-10-10 11:17:15,564 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-10-10 11:17:15,694 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-10-10 11:17:17,262 - octoprint.server - INFO - Shutting down intermediary server...
2018-10-10 11:17:20,384 - octoprint.server - INFO - Intermediary server shut down
2018-10-10 11:17:20,404 - octoprint.events - INFO - Processing startup event, this is our first event
2018-10-10 11:17:20,410 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-10-10 11:17:20,421 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-10-10 11:17:20,573 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-10-10 11:17:20,694 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2018-10-10 11:17:20,765 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2018-10-10 11:17:20,786 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2018-10-10 11:17:20,788 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2018-10-10 11:17:20,945 - octoprint.plugins.pluginmanager - INFO - Looks like we are offline, can't fetch repository from network
2018-10-10 11:17:21,000 - octoprint.plugins.pluginmanager - INFO - Looks like we are offline, can't fetch notices from network
2018-10-10 11:17:21,067 - octoprint.plugins.announcements - INFO - Looks like we are offline, can't fetch announcements for channel _important from network
2018-10-10 11:17:21,116 - octoprint.plugins.announcements - INFO - Looks like we are offline, can't fetch announcements for channel _octopi from network
2018-10-10 11:17:21,407 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.2.36/'}
2018-10-10 11:17:21,575 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 183.37ms
2018-10-10 11:17:21,800 - tornado.access - WARNING - 409 GET /api/printer (127.0.0.1) 187.59ms

#### Link to contents of terminal tab or serial.log

-

#### Link to contents of Javascript console in the browser

[Info] Initial application setup done, connecting to server... (packed_core.js, line 13738)

#### Screenshot(s)/video(s) showing the problem:

-

Yes, I have read the FAQ.

This issue seems to have "piled up". I got server timeouts eventually in the past. But as of today it really got so messy that the server restart did not help.

foosel, 2018-10-15:
Please provide a full `octoprint.log`, not just an excerpt.

And please try to start in [safe mode](http://docs.octoprint.org/en/master/features/safemode.html) (the Contribution Guidelines you were referred to in the bug template also tell you how to do this) and see if the issue persists. Also make sure to delete your browser's cache, it might be something in there causing issues.

As a side note, based on your log excerpt you have OctoPrint 1.3.5. Current version is 1.3.9, released on July 25th. 1.3.5 will be a year old tomorrow. So as a final note, you might also want to try a [manual update](https://discourse.octoprint.org/t/how-can-i-update-the-octoprint-installation-on-my-octopi-image/207).

thomaskilian, 2018-10-15:
FWIW: After some rain dances the issue disappeared (but I had it more than once in the past, just that I was not able to spot the reason). Octoprint tells me it's up to date:

[image]

Anyhow, I'll try that manual update and see what will happen.

foosel, 2018-10-15:
It tells you you are up to date because it looks like it can't connect to the internet to check for a new version (note the broken link icon).

thomaskilian, 2018-10-15:
Yes, now I notice that icon. Strangely enough, this machine and the Pi are in the same network. And clearly I can reach any net address outside the world from here and the Pi.

foosel, 2018-10-15:
Check/provide the *full* log ;)

thomaskilian, 2018-10-15:
Well, my current print has just finished. So I tried this:

> pi@octopi:~ $ ~/oprint/bin/pip install https://get.octoprint.org/latest
Traceback (most recent call last):
  File "/home/pi/oprint/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/home/pi/oprint/lib/python2.7/locale.py", line 579, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting

Also the log (plus the one from yesterday that lists the above mentioned lines). 

[octoprint.log](https://github.com/foosel/OctoPrint/files/2478405/octoprint.log)

[octoprint.2018-10-13.log](https://github.com/foosel/OctoPrint/files/2478416/octoprint.2018-10-13.log)

foosel, 2018-10-15:
Looks like something in your environment is a bit off and hiccuping, making `pip` fall on its face. Try

```
export LC_ALL=C
~/oprint/bin/pip install https://get.octoprint.org/latest
```

([Source](https://stackoverflow.com/a/36394262))

thomaskilian, 2018-10-15:
That seems to do it. Now updating. No idea why the hick-ups. Probably due to "pull the power plug" too often :-/

foosel, 2018-10-15:
> Probably due to "pull the power plug" too often :-/

Yeah... *never* do that. Can break a ton of things and is hard to impossible to fix completely again outside of reimaging.

thomaskilian, 2018-10-15:
Well, sometimes you can't avoid it. The mentioned loop (probably) just hung that tiny machine to that was the last resort.

thomaskilian, 2018-10-15:
It does not look like the installation went through:

[install.log](https://github.com/foosel/OctoPrint/files/2478570/install.log)

thomaskilian, 2018-10-15:
Never mind. I'll install a new image.

foosel, 2018-10-18:
Ok, closing this then.

closed 2018-10-18 by foosel