OctoPrint/OctoPrint #2843: Repeated Communication error with Prusa Mk3 Firmware: 3.4.1-1356

asterixorobelix, 2018-10-10:
#### What were you doing?
I simply connect to printer via octoprint.local and after about 2 minutes receive a: Communication error

Steps to reproduce:
-->
1. login to octoprint.local with credentials
2. Select --> Serial Port: Auto, Baudrate: Auto, Save Connection Settings: Tick, Auto-connect on server startup: Tick
3. Click connect
4. Octoprint connects to printer successfully
5. After about 2 minutes, Octoprint disconnects and displays red Communication Error - State: Offline (Error: Too many consecutive timeouts, printer still connected and alive?)
6. Terminal Output: Attached below

#### What did you expect to happen?
I expected that Octoprint would be able to maintain connection with printer
#### What happened instead?
Octoprint lost connection with printer
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

bolsoncerrado, 2018-10-11:
I think Prusa is aware of this bug on 3.4.1 - you should flash back to a previous firm version.

asterixorobelix, 2018-10-11:
Hi bolsoncerrado

According to the Prusa Changelog (
https://www.prusa3d.com/drivers/firmware-changelog/#mk3), firmware version
3.4.1 should have fixed Octoprint communication issues from their side.

However, I will try downgrading the firmware.

Regards,
Nathan Stasin

bolsoncerrado, 2018-10-11:
I dont think thats the issue they fixed....I think they introduced a new one actually. At least I run all my Prusas with Pi's and never had a communications issue before, until upgrading to 3.4.1....

asterixorobelix, 2018-10-11:
OK, thanks. I will definitely downgrade the firmware then.
Whats the highest firmware version which you have found plays nicely with
octoprint? Did 3.3.1 work for you?

bolsoncerrado, 2018-10-11:
3.4.0 works nice on my Mk3
3.3.1 on my Mk2.5

asterixorobelix, 2018-10-11:
@bolsoncerrado I have downgraded the firmware all the way to 3.1.1 and the problem still occurs

asterixorobelix, 2018-10-11:
Updated Octoprint log file, with downgraded firmware versions:
https://pastebin.com/q50tYUFt

I will see if getting the Prusa people involved adds any insight

asterixorobelix, 2018-10-11:
I will add that I also cant connect to the printer via Pronterface on the mac

asterixorobelix, 2018-10-11:
I have contacted Prusa Technical support who say that they will look into it

foosel, 2018-10-15:
> I will add that I also cant connect to the printer via Pronterface on the mac

If Pronterface also doesn't connect, that definitely sounds like a firmware issue. And looking at the log it shows that the printer simply doesn't respond to OctoPrint's desperate pleas for starting up a dialogue:

```
Send: N0 M110 N0*125
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N1 M105*38
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N0 M110 N0*125
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N1 M115*39
```

It should  have responded with an `ok` to *any* of those commands. But it didn't respond at all.

So - nothing that can be fixed in OctoPrint, needs to be fixed in the firmware. Might be that it hangs on bootup, I've seen this happen occasionally.

closed 2018-10-15 by foosel

suprechent, 2018-11-23:
I had your exact issue.  I just fix mine by turning "RPi" off in settings and everything works as its suppose too.  Hope this helps

aderusha, 2019-01-24:
I had installed an RPi zero W into my new Mk3 but the performance was horrible so I switched back to an RPi3B+ and was seeing this issue.  @suprechent pointed out the source of my problem - I still had the rpi port enabled on the printer.  Disabling that from the printer's LCD panel results in everything working like I expect.  Thanks!

davefuentes, 2019-04-23:
> I had installed an RPi zero W into my new Mk3 but the performance was horrible so I switched back to an RPi3B+ and was seeing this issue. @suprechent pointed out the source of my problem - I still [...]

Hi @aderusha could you please explain how did you achieve that? Is a firmware configuration or Octoprint? Cannot find anything related to "rpi" in either Marlin code or OctoPrint settings.

Thanks a lot!

PS: I'm using Marling 1.1.9 and Octoprint 1.3.10

aderusha, 2019-04-23:
> Hi @aderusha could you please explain how did you achieve that? Is a firmware configuration or Octoprint? Cannot find anything related to "rpi" in either Marlin code or OctoPrint settings.
>
> Thanks a [...]

I'm using the Prusa fork of Marlin so I'm not sure where it lives outside of Prusa's firmware.  For Prusa, it's  under Settings > RPi Port