OctoPrint/OctoPrint #2852: Zeroconf accouncement fails for second instance of OctoPrint

probonopd, 2018-10-13:
#### What were you doing?

Enable Zeroconf announcements:

```
sudo apt -y install git python-virtualenv python-pip squashfs-tools virtualenv python2.7-dev libyaml-dev libavahi-compat-libdnssd-dev
git clone https://github.com/foosel/OctoPrint
cd OctoPrint
virtualenv -p python2.7 venv
./venv/bin/python setup.py install
./venv/bin/python ./venv/bin/octoprint
./venv/bin/pip install https://goo.gl/SxQZ06 # for Avahi; http://docs.octoprint.org/en/master/bundledplugins/discovery.html
```

Then run two instances of OctoPrint:

```
runuser -l user -c '/mnt/bin/python /mnt/bin/octoprint' &
runuser -l user -c '/mnt/bin/python /mnt/bin/octoprint --port=5001 --basedir $HOME/.octoprint2/' & # Second instance
```

#### What did you expect to happen?

OctoPrint announces multiple instances of itself on the network using Zeroconf.

#### What happened instead?

```
2018-10-13 18:40:48,117 - octoprint.plugin - ERROR - Error while calling plugin discovery
Traceback (most recent call last):
  File "/root/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/root/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/discovery/__init__.py", line 143, in on_startup
    self.zeroconf_register("_http._tcp", self.get_instance_name(), txt_record=self._create_http_txt_record_dict())
  File "/root/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.9-py2.7.egg/octoprint/plugins/discovery/__init__.py", line 197, in zeroconf_register
    self._sd_refs[key] = pybonjour.DNSServiceRegister(**params)
  File "/mnt/local/lib/python2.7/site-packages/pybonjour.py", line 1132, in DNSServiceRegister
    None)
  File "/mnt/local/lib/python2.7/site-packages/pybonjour.py", line 286, in _errcheck
    raise cls(result)
BonjourError: (-65548, 'name conflict')
```

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

```
root@vegas805:~# cat /etc/os-release 
NAME="Ubuntu"
VERSION="16.04.4 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.4 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

root@vegas805:~# cat /etc/armbian-image-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=vegas805
BOARD_NAME="S805"
BOARDFAMILY=vegas805
VERSION=5.44
LINUXFAMILY=vegas805
BRANCH=default
ARCH=arm
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=uImage
f33be5cb-cba4-40f4-95cc-85c8393e2142
44f9c833-6d9b-490e-9824-793805583b14
```

#### Printer model & used firmware incl. version

Any

#### Browser and version of browser, operating system running browser

Any

foosel, 2018-10-19:
Ok, this doesn't seem to reproduce under Window, but under Linux it does. That's gonna be a fun one...

foosel, 2018-10-19:
This should now be fixed by the above commit, at least I can no longer reproduce the behaviour and get now the identical one that is available on Windows out of the box (appears to be a difference on the library level). 

Commit is pushed to `maintenance`, soon `devel`, will be part of 1.3.10.

commit 36b07b23 (2018-10-19): discovery: enumerate zeroconf service names on collision

Fixes #2852
closed 2018-12-10 by foosel