OctoPrint/OctoPrint #2853: It's not possible to search or print the documentation

zbrozek, 2018-10-16:
Try using ctrl-f on this page:
https://discourse.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337
The search is intercepted by the page and *then does not work*.

Or try printing it to PDF so you can search the page with a civilized viewer. You only get a little bit of content then it trails off.

This makes using the documentation very difficult.

dforsi, 2018-10-16:
> The search is intercepted by the page and then does not work.

works for me (tested with Chrome 69 and Firefox 52 on Linux)

foosel, 2018-10-16:
Please do not abuse the bug tracker of OctoPrint for reporting issues you encounter while using the community forum. There's a category right in the forum for that.

closed 2018-10-16 by foosel