OctoPrint/OctoPrint #2854: octoprint.util.comm - INFO - Externally triggered heatup detected in endless loop

thomaskilian, 2018-10-16:
#### What were you doing?

Nothing special, except for printing something.

#### What did you expect to happen?

Start log, some messages and final.

#### What happened instead?

Above message kept added to the log in loop.

#### Did the same happen when running OctoPrint in safe mode?

yes

#### Version of OctoPrint

OctoPrint 1.3.9 running on OctoPi 0.15.1

#### Operating System running OctoPrint

see above

#### Printer model & used firmware incl. version

DIY running Teacup. I guess the latter could be the culprit.

#### Browser and version of browser, operating system running browser

irrelevant

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/2483928/octoprint.log)

#### Link to contents of terminal tab or serial.log

could b provided if needed...

#### Link to contents of Javascript console in the browser

irrelevant

#### Screenshot(s)/video(s) showing the problem:

irrelevant

Yes, I have read the FAQ.

foosel, 2018-10-16:
> could b provided if needed...

Always needed when it comes to printer interaction, please provide

thomaskilian, 2018-10-16:
[octoprint.log](https://github.com/foosel/OctoPrint/files/2485111/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/2485112/serial.log)

foosel, 2018-10-18:
Yep, looks indeed like the firmware is the culprit here. The temperature message is a bit more Repetier than Marlin like, but there's a simple solution to make things not endlessly log that offending log line.

Settings > Serial Connection > Firmware & protocol, there uncheck "Enable automatic firmware detection" and manually check "Disable detection of external heatups". That should solve it - please report back on that.

Also, if you could provide me the output of `M115` (or just a `serial.log` from a **freshly started connection** to the printer) I might be able to have this handled correctly automatically in future versions.

thomaskilian, 2018-10-18:
Yes, will do. I have a print job currently running...

thomaskilian, 2018-10-21:
Now's the time. However, over night while the above log lines piled up I got now additional loads of

> 2018-10-20 18:27:40,704 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.2.48) 873.18ms
2018-10-20 18:27:42,073 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.2.48) 1026.31ms
2018-10-20 18:27:43,280 - tornado.access - WARNING - 403 GET /api/settings (::ffff:192.168.2.48) 961.32ms

Will now restart, change settings and see what the next print comes up with.

thomaskilian, 2018-10-21:
Hmm. I only restarted the service and the latest mentioned log files keep flooding. Reboot it must be - I guess.

thomaskilian, 2018-10-21:
So, reboot could not get rid of it: 
[octoprint.log](https://github.com/foosel/OctoPrint/files/2498973/octoprint.log)

Taken tight after a reboot with the settings you proposed above done before.

thomaskilian, 2018-10-21:
Aaaaaah!. I had a Cura instance open. Once I closed that the issue went away and OctoPrint is back to work. Probably it's gone wild trying to communicate with OctoPrint.

foosel, 2018-10-22:
So... I'm confused, is the problem now gone or not? :)

thomaskilian, 2018-10-22:
Just to confirm: the issue (now) was Cura, not OctoPrint.

foosel, 2018-10-22:
I was referring to the initially reported problem of looping "external heatup detected" messages in the log for which I suggested a workaround (and asked for some more info to detect this printer out of the box) [here](https://github.com/foosel/OctoPrint/issues/2854#issuecomment-431035889).

thomaskilian, 2018-10-22:
Oh. Seems I lost that out of sight :-( Yes. The original issue went away with your receipt.

foosel, 2018-10-22:
Could you provide me with your printer's output to `M115`/a `serial.log` of a *fresh connection* to your printer then? I might be able to add it to the auto detection then so others who own it don't run into the same issue.

thomaskilian, 2018-10-22:
Sure. Will probably be Wednesday then.

thomaskilian, 2018-10-28:
output from M115:

> Recv: FIRMWARE_NAME:Teacup FIRMWARE_URL:http://github.com/traumflug/Teacup_Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 TEMP_SENSOR_COUNT:2 HEATER_COUNT:3

And a dummy print:
[serial.log](https://github.com/foosel/OctoPrint/files/2522903/serial.log)

foosel, 2018-10-29:
Perfect! With that I could add a new line to the firmware auto detection and in the future the corresponding flag will be set automatically for teacup firmware.

Already pushed to `maintenance`, soon `devel`, to be released as part of 1.3.1.0.

thomaskilian, 2018-10-29:
Thank you! I'll test that as soon as time permits.

thomaskilian, 2018-11-13:
I just tested with 1.3.10rc2 and it seems like it's working :-)

foosel, 2018-12-10:
1.3.10 has been released.

closed 2018-12-10 by foosel